NatLabRockies/ssc #24: Implementation of function for module spectral mismatch considering precipitable water

trichrt, 2017-08-29:
The _PVPerformance Modelling Collaborative_ defines two equations that can be used to describe the spectral behavior of a PV module:

1. Based on the **Sandia PV Array Model**: A 4th degree polynomial with five coefficients defining a spectral mismatch modifier as a function of Air Mass.
2. Based on **research by Lee/Panchula** from First Solar: A function with six coefficients defining spectral mismatch as a function of air mass and precipitable water.

To my knowledge, only the first function is currently implemented and used in SAM/SSC in several modules. My suggestion would be to implement the second function as well and make it available as a user input at least in the IEC61853 model. 

Related links:
https://pvpmc.sandia.gov/modeling-steps/2-dc-module-iv/effective-irradiance/spectral-mismatch-models/
http://ieeexplore.ieee.org/document/7749836/

janinefreeman, 2017-08-29:
This would be a valuable addition that has a high likelihood of being included in the official NREL version of SAM, if implemented robustly! The Lee/Panchula model provides more detail, without requiring too much additional weather file input information. One implementation consideration is whether to always calculate precipitable water from temperature and relative humidity in the weather file, or to also allow the option for the user to input precipitable water directly in the weather file. The latter would likely require some modifications to the weather file reading routines.

Related to adding a new spectral model: Each module model  (CEC, Sandia, Simple Efficiency) currently has its own way of treating the spectral behavior of a PV module. It would be great to also separate the spectral models from the module models, so that any spectral model can be used with any module model. This would require some re-working of the user interface, as well as significant code reorganization. There is a possibility the NREL team would try to tackle this piece of it in the next year or so.

If someone is interested in tackling this new feature, please contact us so that we can coordinate and collaborate! sam.support@nrel.gov

trichrt, 2017-08-29:
> It would be great to also separate the spectral models from the module models, so that any spectral model can be used with any module model.

I agree, that would make absolute sense. As far as I see in the code, there is the central function `air_mass_modifier`, used to calculate the modifier in the `lib_cec6par.cpp`, `cmod_hcpv.cpp`, and `lib_iec61853.cpp` module models. At least on the SSC-side this function could thus be extended easily. 

However, elaborating how to separate the module models from the spectral functions would require some architectural thought first and definitely be more work than changing the function itself.

A straightforward way would be to add the new six input parameters as well as a switch parameter (to choose the applied equation) to each module input parameter list. But there are probably smarter ways to do this!?