NatLabRockies/ssc #26: Implementation of the Mermoud/Lejeune module model

timorichert, 2017-08-31:
Due to its wide application in the solar industry and its use as the de-facto standard in bankable yield assessment of solar power plants, the **Mermoud/Lejeune module model** could be a valuable addition for SAM/SSC.

The governing equations are published in ["Performance assessment of a simulation model for PV modules of any available technology", André Mermoud  and Thibault Lejeune, 2010](http://www.pvsyst.com/images/papers/thf_article_valence_2010.pdf). A summary can be found on the [PVPMC website](https://pvpmc.sandia.gov/modeling-steps/2-dc-module-iv/diode-equivalent-circuit-models/pvsyst-module-model/) as well.

One advantage is that module manufacturers typically make their module data available in the PAN format, which contains parameters fit for use with this module models' governing equations. Examples for PAN files can be found for [First Solar modules](http://www.firstsolar.com/en-EMEA/Resources/Technical-Documents?ty=&to=&q=&page=1) for the older PAN format or from [Astroenergy](http://www.astronergy-solarmodule.de/de/download.html) for the more recent format. Thus, implementing this model would allow to import and use manufacturer data easily in SAM.

The model has already been implemented in the [CanadianSolar/CASSYS](https://github.com/CanadianSolar/CASSYS) open source simulation tool in C#. This code could serve as a reference in the development. Its licensing terms can be found [here](https://github.com/CanadianSolar/CASSYS/blob/master/LICENSE). On the other hand, I assume that the existing SAM code base could be very easily adapted to use these governing equations in addition to the ones currently implemented.

timorichert, 2017-09-03:
FYI: I have started to implement the governing equations in a new and separate module model that can be used next to the other modules types in the SSC. You can find the code in the [`pvyield/ssc`](https://github.com/pvyield/ssc) fork in commit 2b18e02006ad57011047e72829fa4dd7cb440459. The code is derived from the [lib_cec6par-model](https://github.com/NREL/ssc/blob/develop/shared/lib_cec6par.cpp) and makes use of the already implemented SSC-functionality of determining V_oc and P_mp. The model will generally allow the consideration of different types of IAM-definition, AirMass-Modifiers, temperature models etc., even though not all options have been implemented at this point.

The code compiles well for x64 systems, the input parameters can be viewed in the SDKtool as they should, and some hourly simulation runs done for testing work fine. The results of the equations have been checked in Debug mode for some states and they agree with a benchmark model setup in Excel in parallel.

**Still, I advise that this is beta code that needs more thorough testing before live use.**

In case there is interest in merging this functionality into the SSC main branch I would create a pull request. Any advisable changes and tasks to be performed before the implementation (i.e. change of variable names, change of the model prefix, tests to be run before merging the code, etc.) could be discussed there.

cwhanse, 2018-07-27:
It doesn't appear that this code is merged yet.  The current version of the PVsyst model is different from that described in the Mermoud 2010 paper, specifically, the equation for Rsh.  The description at [PVPMC](https://pvpmc.sandia.gov/modeling-steps/2-dc-module-iv/diode-equivalent-circuit-models/pvsyst-module-model/) is up to date.

janinefreeman, 2018-07-27:
We are planning to merge this code in the next month or two. The model will not immediately be made available in the UI, but will be available via the SDK.

cwhanse, 2018-07-27:
I'd recommend that the equations for Rsh be updated to PVsyst v6.  The difference is on this line: 

`R_sh = R_shref + (R_sh0 - R_shref) * exp(-R_shexp * (S / S_ref));`

should be (I'm sure I'll get the syntax wrong)

`R_shbase = max( (Rsh_ref - R_sh0 * exp(-R_shexp) / (1 - exp(-R_shexp)), 0);`
`R_sh = R_shbase + (R_sh0 - R_shref) * exp(-R_shexp * (S / S_ref));`

timorichert, 2018-10-24:
> It doesn't appear that this code is merged yet. The current version of the PVsyst model is different from that described in the Mermoud 2010 paper and coded in these commits, specifically, the equation for Rsh. The description at [PVPMC](https://pvpmc.sandia.gov/modeling-steps/2-dc-module-iv/diode-equivalent-circuit-models/pvsyst-module-model/) is up to date.

Thanks for checking and for the insight, @cwhanse!

timorichert, 2018-10-24:
> Working in the release branch with spline.cpp. Soon to be replaced with SPLINTER for cubic spline implementation

Great, @sjanzou, thanks for finding a license-compatible spline interpolator!

sjanzou, 2018-10-27:
Completed and tests set up in release branch

closed 2018-10-27 by sjanzou

allenlawrence94, 2021-08-06:
Is this model able to model bifacial modules?

janinefreeman, 2021-08-06:
Unfortunately this model currently doesn't have a bifacial option implemented, although it is a wishlist item that has been noted in #241 . This model was contributed externally and is currently available through the SDK/PySAM only. At one point @caseyzak24  was interested in adding the bifaciality capability to this model, although I don't know if he's had time to do so. If any external contributors were interested in adding that functionality, we'd be happy to help provide guidance as we know lots of users would appreciate it!

allenlawrence94, 2021-10-05:
@janinefreeman - I adapted one commit from @caseyzak24's work opened a PR here: https://github.com/NREL/ssc/pull/622. This PR simply exposes bifaciality as an option in the mermoud module model. Thanks for your guidance! Let me know if/when this can be reviewed.