NatLabRockies/ssc #51: sCO2 Brayton Power cycle new configurations (reheating, PCRC, RCMCI).

luiscoco, 2017-10-06:
Dear developers,
Based on the files sco2_pc_core.h, sco2_pc_core.cpp, I started programming the recompression power cycle with one ReHeating stage in the turbine. 
My plan is after integrating the ReHeating in the recompression power cycle continuing programming the PCRC, RCMCI cycles with reheating and without reheating.
Please confirm me you agree with my developing plan, and if these new capabilities are beneficial for central tower, parabolic and fresnel thermosolar power plants modelled in SAM.
Best Regards,
Luis Coco Enríquez 
Contact e-mail: LUISCOCOENRIQUEZ@HOTMAIL.COM

luiscoco, 2017-10-17:
[Recompression_with_ReHeating.zip](https://github.com/NREL/ssc/files/1391946/Recompression_with_ReHeating.zip)

luiscoco, 2017-10-17:
Dear Nick, 
Find attached my progress for adding the ReHeating capability to the Recompression power cycle.
I continue with my work this week modifying the sco2_pc_csp_int.h and sco2_pc_csp_int.cpp files. 
Best Regards,
Luis Coco Enríquez

dguittet, 2020-09-24:
closing for inactivity

closed 2020-09-24 by dguittet