NatLabRockies/ssc #80: Unable to build SDKtool

bje-, 2017-11-07:
This has never worked for me in the Linux build. Am I doing something wrong?

    make[1]: Entering directory '/home/bje/source/ssc.bje/build_linux'
    g++ -g -o SDKtool dataview.o dllinvoke.o editvariable.o scripting.o sscdev.o ../../wex/wexuxwx3.a -std=c++0x ../../wex/wexuxwx3.a ../../lk/lkuxwx3.a `wx-config-3 --libs stc` `wx-config-3 --libs aui` `wx-config-3 --libs` -lm  -lfontconfig -ldl -lcurl 
    /usr/bin/ld: ../../wex/wexuxwx3.a(pltext.o): undefined reference to symbol 'FT_Library_SetLcdFilter'
    //usr/lib/x86_64-linux-gnu/libfreetype.so.6: error adding symbols: DSO missing from command line

nickdiorio, 2017-11-10:
Hmm, when I do `make` within `ssc/build_linux`, I eventually get to the SDK tool part, which is:

```
make -f Makefile-sdktool -j4
make[1]: Entering directory '/local/workspace/ssc/build_linux'
g++ -std=c++0x -g -O2 -I. -I../../wex/include -I../../lk/include -I../ssc -I../shared -DLK_USE_WXWIDGETS `wx-config-3 --cflags` -Wall -Wno-unknown-pragmas   -c -o dataview.o ../sdktool/dataview.cpp
g++ -g -o SDKtool dataview.o dllinvoke.o editvariable.o scripting.o sscdev.o ../../wex/wexuxwx3.a -std=c++0x ../../wex/wexuxwx3.a ../../lk/lkuxwx3.a `wx-config-3 --libs stc` `wx-config-3 --libs aui` `wx-config-3 --libs` -lm  -lfontconfig -ldl -lcurl 
make[1]: Leaving directory '/local/workspace/ssc/build_linux'
```

Are you invoking `make` in that way?  Also, the undefined symbols are from `libfreetype`.  To install all necessary dependencies (in Debian/Ubuntu/Mint), you should be able to do:

```
sudo apt-get install -y libgtk2.0-dev libgl1-mesa-dev mesa-common-dev freeglut3-dev libcurl4-openssl-dev libc6
```

bje-, 2017-11-10:
I just installed those packages and a few things were missing, but that didn't seem to help. I'll dig in deeper. It's not normally too difficult to diagnose missing symbol problems given enough time. :-)

bje-, 2017-11-11:
Fixed with NREL/wex#39.

closed 2017-11-11 by bje-