NatLabRockies/ssc #95: query in tcs/sam_mw_trough_type250.cpp

bje-, 2017-12-14:
When compiling `sam_mw_trough_type250.cpp` with clang, the following warnings are produced. `reguess_args` is an array of doubles, so it is never `NULL`. What is intended here?

    ../tcs/sam_mw_trough_type250.cpp:3051:6: warning: comparison of array 'this->reguess_args' equal to a null pointer is always false
          [-Wtautological-pointer-compare]
                    if(reguess_args == NULL) goto lab_reguess;
                       ^~~~~~~~~~~~    ~~~~
    ../tcs/sam_mw_trough_type250.cpp:3090:8: warning: comparison of array 'this->reguess_args' not equal to a null pointer is always true
          [-Wtautological-pointer-compare]
                                    if(reguess_args != NULL){
                                       ^~~~~~~~~~~~    ~~~~
    ../tcs/sam_mw_trough_type250.cpp:3102:8: warning: comparison of array 'this->reguess_args' not equal to a null pointer is always true
          [-Wtautological-pointer-compare]
                                    if(reguess_args != NULL){
                                       ^~~~~~~~~~~~    ~~~~

nickdiorio, 2017-12-15:
@tyneises , do you have any idea what was intended here?  It doesn't appear the goto ever gets hit here

bje-, 2017-12-18:
I propose just deleting these lines for now. As it stands, they do nothing.

tyneises, 2017-12-18:
cleaned up a bit here: https://github.com/NREL/ssc/commit/bf66a814de92437c29b3579ccca6eb5743f84705

closed 2017-12-18 by tyneises