NatLabRockies/ssc #96: query in tcs/csp_solver_pt_sf_perf_interp.cpp

bje-, 2017-12-14:
clang reports the following warning when compiling `tcs/csp_solver_pt_sf_perf_interp.cpp`:

    ../tcs/csp_solver_pt_sf_perf_interp.cpp:373:16: warning: assigning field to itself [-Wself-assign-field]
            m_v_wind_prev = m_v_wind_prev;

What was intended here, or can this statement be deleted?

nickdiorio, 2017-12-15:
It's difficult for me to see how these are anything but trivial assignments.  I suspect they are mostly for commenting purposes.  Feel free to remove in any subsequent pull requests, thanks!

closed 2017-12-15 by nickdiorio

bje-, 2017-12-16:
Thanks. Inspecting the code in storage_hx.cpp shows what the real intent was -- to set a class member based on a function argument just like the 'cold' case in the line above. See the forthcoming PR for the details.