NatLabRockies/ssc #129: Module coefficient generator 6parsolve does not work for First Solar Series 6 CdTe modules.

cpaulgilman, 2018-02-22:
https://sam.nrel.gov/node/75678

janinefreeman, 2018-02-27:
Fails within "solve" function in 6par_solve.h (line 466). In UI, both in released version and open source version (built today), changing celltype input to "monoSi" or "multiSi" causes it to solve correctly. In SDKtool, I can't replicate that- monoSi and multiSi both fail, built in both 32-bit and 64-bit debug.  SDKtool script follows. Still working.

	//calculate inputs
	celltype= 'monoSi';//['monoSi', 'multiSi/polySi', 'cdte', 'cis', 'cigs', 'amorphous'];
	voc = 215;
	isc = 2.63;
	
	//call 6parsolve module
	var('celltype', celltype);
	var('Vmp', 179.4);
	var('Imp', 2.37);
	var('Voc', voc);
	var('Isc', isc);
	var('alpha_isc', -0.28 * 0.01 * isc);
	var('beta_voc', 0.04 * 0.01 * voc);
	var('gamma_pmp', -0.32);
	var('Nser', 216);
	var('Tref', 25);
	result= run('6parsolve');

janinefreeman, 2018-03-01:
With the given specifications, SAM fails to solve for 5-parameter coefficients when "cdte" is specified as the technology, but succeed when selecting "monoSi" or "multiSi". This input only affects the initial guesses for an iterative solver. More investigation needed as to the source of these initial guesses, but it would seem that current workaround of using monoSi or multiSi, even though it is a CdTe module, is a legitimate option.

dguittet, 2018-06-11:
see comment in pull request for fix details

cpaulgilman, 2020-09-11:
Coefficient generator is not solving for some newer modules, for example parameters from these Trina modules marked "preliminary" :  [6parsolve-convergence-issues.zip](https://github.com/NREL/ssc/files/5211782/6parsolve-convergence-issues.zip). (These are bifacial modules, solver is only considering front side of module.)

CSV file in zip files lists module parameters and initial test results. Use lk scripts to read parameters from CSV file and run 6parsolve.

Solver does work for these similar Trina modules: 
[EN_Datasheet_Duomax_Twin_DEG15HC.20II_201907.pdf](https://github.com/NREL/ssc/files/5211783/EN_Datasheet_Duomax_Twin_DEG15HC.20II_201907.pdf)
[MA_Datasheet_Vertex_DEG18MC.20(II)_2020A.pdf](https://github.com/NREL/ssc/files/5211784/MA_Datasheet_Vertex_DEG18MC.20.II._2020A.pdf)

For Solibro module in zip file, changing celltype from cigs to multiSi causes solver to solve.

PR #172 (2018-06-11): Fix module coefficient generator for CdTe modules
The numerical [solver](http://www.aip.de/groups/soe/local/numres/bookcpdf/c9-7.pdf) (globally convergent newton method) is very sensitive to initial guesses. Effect on convergence due to differing initial guesses for each of the 6 parameters and uniqueness of convergent solution was explored over the search space. 

Due to large difference in magnitude of the parameters (1e-17 to 1e3), rescaling of the problem to unity was attempted but did not have any effect on convergence.

Since convergence was found to be dependent primarily on initial gue [... cut]
commit 97ba8195 (2018-06-11): fix #129 module coefficient generator for CdTe
commit dc881d51 (2020-11-07): close #129
commit c1a40c53 (2020-11-09): Update 6par Module Model Parameter Generator (#480)

* add failing 6par_solve tests for new monoSi modules

* close #129

* Fix "and" typo

* update 6par_solve_test
closed 2020-11-09 by dguittet