NatLabRockies/ssc #130: modules for hydrogen electrolysis

yassineennassiri, 2018-02-26:
Hi ! I would like to add new modules to SSC library in ordre to build model for production of hydrogen using eletrolysis (The eletricity needed in electrolysis of water will be generated from PV/CSP system). I'm not sure how to get started, so wanted to touch base with your team to plan how I can add this new feature.

nickdiorio, 2018-02-27:
Hi @yassineennassiri , thank you for your interest! Our team will discuss this and see how if might fit in SSC.  Just to clarify, are you interested in a combined model of PV/CSP + Hydrogen, or a generic model where you simply input the electricity profile from a generic source and then run the hydrogen model?

yassineennassiri, 2018-02-28:
Hi !

Actually, I'm interested in a combined model of PV/CSP + hydrogen.

Thanks for your support!

<https://github.com/NREL/ssc/issues/130#issuecomment-369068089>, or mute
the thread

nickdiorio, 2018-03-13:
Hi @yassineennassiri , sorry for the delay.  We've been talking over your request internally, as well as developing a more robust method to plan large contributions.  We're finalizing some things on our end, and then I'll reach out to you to ask you to provide a detailed plan for what you need and how you will approach it.  Thanks again.