NatLabRockies/ssc #140: Bug in subhourly EPW file reading

nickdiorio, 2018-03-09:
A user has encountered problems with SAM being unable to handle subhourly EPW files. If using the PVWatts option, the simulation will misinterpret a 15-min file as hourly and only use the first 25% of the inputs. If using the Detailed PV option, the user gets an error message, and the simulation fails.

nickdiorio, 2018-03-09:
The recommended work around for this issue in the immediate term is to convert the file to a SAM CSV file

janinefreeman, 2018-08-23:
Proposed path forward: error out in all compute modules if EPW is subhourly, and make sure that the conversion macro to SAM CSV properly handles subhourly EPW files.

dguittet, 2018-09-13:
Added support for subhourly EPW files and tested on 30 min file. 
https://github.com/NREL/ssc/commit/2a261a0589bcbe12fc7aae5b3ab12d1b23660a29

closed 2018-09-13 by dguittet