NatLabRockies/ssc #151: Calculate and report clear sky DNI

mjwagner2, 2018-03-22:
The clear-sky direct normal irradiance (DNI) profile is sometimes useful for CSP analysis. There are several models available for estimating this value, which is location dependent and can account for certain exogenous factors, such as current humidity, pressure, or seasonal effects. The goal of calculating clear-sky DNI is typically to serve as a comparison to observed DNI, and to assist in determining the fraction of DNI energy lost over each time period due to cloudiness. In this respect, clear-sky models that predict what the DNI _would have been_ had there been no cloudiness during a time period would be most useful.

The SolarPILOT code already contains several clear-sky models, taken from 
>Kistler, B.L. “A User’s Manual for DELSOL3: A Computer Code for Calculating the Optical Performance and Optimal System Design for Solar Thermal Central Receiver Plants.” Albuquerque, NM: Sandia National Laboratory, 1986.

These include:
* Meinel 
[image]
* Hottel
[image]
* Allen
[image]
[image]
* Moon
[image]

----

All of these models are fairly old, and newer options are available, e.g. the Perez & Ineichen formulation implemented in [pvlib](https://github.com/pvlib/pvlib-python).

janinefreeman, 2018-03-22:
Mike, we've been wanting clear-sky info in the PV models as well. Maybe there's an easy way to make it available for multiple models (irradproc?) and add more info.

mjwagner2, 2018-03-22:
That seems like a good place for it. It might be a good idea to code up a few models and allow the user to flag which should be used. I'm not sure what the best default model should be, though.

cpaulgilman, 2021-04-22:
A user request for this feature: https://sam.nrel.gov/forum/forum-general/3410

cpaulgilman, 2021-05-06:
This issue is partially addressed in https://github.com/NREL/SAM/pull/609.

Clear sky DHI, DNI, and GHI data is available from the NSRDB. As of this pull request, SAM downloads the clear sky data so it is available in the weather file along with the surface irradiance values. To make this more useful, we should consider having SAM's solar  models (CSP, SWH, PV) read the data and pass it through to show in results with documentation in Help explaining that the data is not used in simulations.

mjprilliman, 2023-06-15:
@cpaulgilman @janinefreeman This is relevant again for the subhourly clipping work. I'm currently spending some time looking into potential models or options like the NSRDB. One early potential feature request: can we have clearsky irradiance values visible in the DView widget on the Resource tab.

mjprilliman, 2023-11-02:
Closing with https://github.com/NREL/ssc/pull/1050, https://github.com/NREL/ssc/pull/1077, see https://github.com/NREL/SAM/issues/1449 for future work.

closed 2023-11-02 by mjprilliman