NatLabRockies/ssc #153: weatherreader issues

jdpipe, 2018-04-24:
While attempting to make some modifications to SSC, I stumbled into some issues with compilation owing to the fact that global variables `weatherreader` and `weatherreader_variables` are defined separately in two difference places. The first is in `tcs/weatherreader.cpp` and the second is in `tcs/weatherreader_csp_solver.cpp`.

These two files both seem to be getting compiled and linked. But I wondered why there need to be two classes with the same name, and global variables with the same name? And in `tcs/typelib.cpp`, which of the two is being referred to? 

I note that various cmod_* files reference `"weatherreader"` via calls to `add_unit`. So how is this working, is this type control system stuff superseding one `weatherreader` with another somehow?

mjwagner2, 2018-04-24:
It appears that weatherreader.cpp is not linked on Visual Studio builds, but is linked on at least Linux. This seems to imply that weatherreader.cpp is no longer needed. I was able to build successfully on Linux without the file. If @tyneises confirms, I suggest that we drop weatherreader.cpp and all references in makefiles.

tyneises, 2018-04-24:
I agree, it looks like we can drop weatherreader.cpp

sjanzou, 2018-04-26:
Fixed by tyneises with commit 073cef04e3bd8969666f1c9eb3341eded29940f9.
Tested iOS, Android, and OSX builds by sjanzou

closed 2018-04-26 by sjanzou