NatLabRockies/ssc #154: utilityrate5

YingkeZhao, 2018-05-03:
1. how could i get utilityrate5 manual?
2.  code bug : double excess_kwhs_earned_ym and excess_kwhs_earned_ym defined.
the rigtht code maybe:

	// added for monthly bill balancing per 12/14/16 meeting
	{ SSC_OUTPUT, SSC_MATRIX, "charge_w_sys_ec_gross_ym", "Energy charge with system before credits", "$", "", "Charges by Month", "*", "", "COL_LABEL=MONTHS,FORMAT_SPEC=CURRENCY,GROUP=UR_AM" },
	{ SSC_OUTPUT, SSC_MATRIX, "excess_dollars_applied_ym", "Excess generation $ credit applied", "$", "", "Charges by Month", "*", "", "COL_LABEL=MONTHS,FORMAT_SPEC=CURRENCY,GROUP=UR_AM" },
	{ SSC_OUTPUT, SSC_MATRIX, "excess_dollars_earned_ym", "Excess generation $ credit earned", "$", "", "Charges by Month", "*", "", "COL_LABEL=MONTHS,FORMAT_SPEC=CURRENCY,GROUP=UR_AM" },
	{ SSC_OUTPUT, SSC_MATRIX, "excess_kwhs_applied_ym", "Excess generation kWh credit applied", "kWh", "", "Charges by Month", "*", "", "COL_LABEL=MONTHS,FORMAT_SPEC=CURRENCY,GROUP=UR_AM" },
	{ SSC_OUTPUT, SSC_MATRIX, "excess_kwhs_earned_ym", "Excess generation kWh credit earned", "kWh", "", "Charges by Month", "*", "", "COL_LABEL=MONTHS,FORMAT_SPEC=CURRENCY,GROUP=UR_AM" },

YingkeZhao, 2018-05-03:
another bug about excess_kwhs_earned_ym ， the right code maybe:
				excess_dollars_applied_ym[(i + 1) * 12 + j] = monthly_excess_dollars_applied[j];
				excess_dollars_earned_ym[(i + 1) * 12 + j] = monthly_excess_dollars_earned[j];
				excess_kwhs_applied_ym[(i + 1) * 12 + j] = monthly_excess_kwhs_applied[j];
				excess_kwhs_earned_ym[(i + 1) * 12 + j] = monthly_excess_kwhs_earned[j];

sjanzou, 2018-10-02:
Thank you for the detailed feedback. Issues noted are fixed.

commit 24ab30b6 (2018-10-02): Update cmod_utilityrate5.cpp - closes #154
closed 2018-10-02 by sjanzou