NatLabRockies/ssc #155: thin film recombination current loss term

mikofski, 2018-05-07:
According to [PVPMC "modeling steps"](https://pvpmc.sandia.gov/modeling-steps/2-dc-module-iv/diode-equivalent-circuit-models/pvsyst-module-model/) and [PVSyst help files](http://files.pvsyst.com/help/thinfilm_recombinationloss.htm) there is a deviation between the single-diode model (SDM) and measurements of current for thin-film devices as voltage approaches open-circuit. A correlation was proposed by J. Merten which is discussed in [Mermoud, Lejeune [2010] **Performance Assessment of a Simulation Model for PV Modules of Any Available Technology** _EUPVSEC-25_](https://github.com/NREL/ssc/files/1981028/Mermoud.Lejeune.-.2010.-.Performance.Assessment.of.a.Simulation.Model.for.Pv.Modules.of.Any.Available.Technology.pdf). See also [**Improved equivalent circuit and analytical modle for a:Si solar cells and modules** J Merten, _et al._ [1998] _IEEE Trans on Elec Dev_](https://doi.org/10.1109/16.658676).

The basic form of the equation is:

    I_rec = I_L * d2_mu_tau / (V_bi - V_diode)

where:

* `d2_mu_tau` is the ratio of the square of the thickness of the intrinsic thin flim layer (d<sub><i>i</i></sub><sup>2</sup>), and the mean effective diffusion length (&mu;&tau;<sub><i>eff</i></sub>), typically given as 1.4V
* `V_bi` is the "built-in" voltage of the intrinsic layer, typically given as 0.9V for a:Si, but in practice is varied to minimize difference between measured and predicted  Voc (or Pmp).
* `V_diode = Vcell + Icell * Rseries`  is the junction or "diode" voltage of the SDM

In practice combine the thin film recombination current `I_rec` using Kirchhoff's law

    I_cell = I_L - I_diode - I_shunt - I_rec

nickdiorio, 2018-05-14:
Thanks @mikofski , we'll take a look at our single-diode model and see how we can appropriately incorporate this correlation.  @janinefreeman , FYI.

nickdiorio, 2018-06-21:
Adding @dguittet

timorichert, 2018-07-26:
Hi @mikofski and @nickdiorio!

I have been working on some adaptions in the SSC code to be able to consider recombination losses as proposed in the initial post. Essentially I introduced three new PV functions in `lib_pvmodel.cpp` to accommodate the additional term:

- `current_5par_rec` (based on `current_5par`): https://github.com/pvyield/ssc/blob/04f61f36614cf409c90d3ccdbfeb22e9a44336f2/shared/lib_pvmodel.cpp#L342-L375
- `openvoltage_5par_rec` (based on `openvoltage_5par`): https://github.com/pvyield/ssc/blob/04f61f36614cf409c90d3ccdbfeb22e9a44336f2/shared/lib_pvmodel.cpp#L403-L428
- `maxpower_5par_rec` (based on `maxpower_5par`): https://github.com/pvyield/ssc/blob/04f61f36614cf409c90d3ccdbfeb22e9a44336f2/shared/lib_pvmodel.cpp#L471-L497

By setting `D2MuTau ` to zero the additional terms are ignored, so these terms could also be added to the original functions, with the two new input variables being assigned zero where not used. For my purpose, it was cleaner to add functions instead of tinkering with existing functionality.

mikofski, 2018-08-01:
@nickdiorio , @timorichert , FYI users will not be able to use existing CEC module coefficients work this Merten _et al._ thin film recombination loss term. See the [plots and comments in pvlib comment](../../../pvlib/pvlib-python/pull/504#issuecomment-408650788). I've just learned from @jdnewmil and @cwhanse so you may want to reconsider whether this is a feature SAM should implement. It would be useful to hear from First Solar and others of there are unacceptable errors modeling CdTe in SAM currently.

timorichert, 2018-08-01:
@mikofski I've developed these additional functions with the recombination loss term for use with a module model that is in line with the [PVsyst module model equations](https://pvpmc.sandia.gov/modeling-steps/2-dc-module-iv/diode-equivalent-circuit-models/pvsyst-module-model/) that you stated in the initial post. I agree that these terms should not be used with regular CEC coefficients (at least not without thorough validation)! However, it should be implemented if a PVsyst-compatible module model will be implemented as I proposed [here](https://github.com/NREL/ssc/issues/26). I have developed such a module model and @janinefreeman and I are currently discussing its implementation in SAM.

nickdiorio, 2018-08-07:
@mikofski , @timorichert  thanks for the updates. We have it on our slate to begin investigating this loss term very soon.  We are aware of some limitations in modeling CdTe in SAM currently (primarily spectral effects), though can definitely reach out to the Plant Predict team to discuss if there are any other large effects we are missing.