NatLabRockies/ssc #173: Incident beam formula issue

npascutti, 2018-06-12:
The formula to calculate the incident beam component of the incident irradiation on a tilted surface seems to be missing the division by cos(zen).
Duffie (John A. Duffie, William A. Beckman - 2013 - Solar Engineering of Thermal Processes, Fourth Edition) reports in equation 2.15.1 (page 89) that for the Isotropic Sky model  the incident beam should be calculated as:

incident beam irradiance = direct normal (beam) irradiance * cos(inc)/cos(zen)

while in line 788 of the code it's calbulated as:

incident beam irradiance = direct normal (beam) irradiance * cos(inc)

nickdiorio, 2018-06-20:
Thanks @npascutti , we will investigate.

nickdiorio, 2018-08-23:
Hi @npascutti , sorry for the delay.  The incident beam calculation in SAM is `dn * cos(inc)` as you mentioned.  I'm not sure why it's divided by the `cos(zen)` in Duffie and Beckman.  One reference for our approach is: https://pvpmc.sandia.gov/modeling-steps/1-weather-design-inputs/plane-of-array-poa-irradiance/calculating-poa-irradiance/poa-beam/ , and here: https://www.osti.gov/pages/servlets/purl/1235343

closed 2018-08-23 by nickdiorio