NatLabRockies/ssc #192: SSC incorrectly refuses to use high-resolution weather data

kevinsa5, 2018-08-23:
**Describe the bug**
When running a pvsamv1 simulation with two years of 1-minute weather data, SSC produces the following error: 
`exec fail(pvsamv1): invalid number of data records (1051200): must be an integer multiple of 8760`.  This is triggered by this line of code in shared/lib_pv_io_manager.cpp: https://github.com/NREL/ssc/blob/develop/shared/lib_pv_io_manager.cpp#L129

```
	//total number of records in the weather file (i.e. 8760 * timestep)
	numberOfWeatherFileRecords = weatherDataProvider->nrecords();
	stepsPerHour = numberOfWeatherFileRecords / 8760;
	dtHour = 1.0 / stepsPerHour;

	if (stepsPerHour < 1 || stepsPerHour > 60 || stepsPerHour * 8760 != numberOfWeatherFileRecords)
		throw compute_module::exec_error(cmName, util::format("invalid number of data records (%zu): must be an integer multiple of 8760", numberOfWeatherFileRecords));
```

As 1051200 is indeed an integer multiple of 8760, I believe that either this error message or the surrounding logic should be revised.  In particular, I believe the issue lies in the line `stepsPerHour = numberOfWeatherFileRecords / 8760;`, which does not actually calculate steps per hour for weather files covering more than one year -- eg a 2-year simulation with 5-minute weather data will calculate 24 steps per hour, not 12.  Note that if SSC requires weatherfiles to cover only a single year, then this check will still permit the 2-year, 5-minute data example.

kevinsa5, 2018-08-23:
Hey @janinefreeman , thanks for the reply!  I did end up getting around this issue by running two years separately and combining the output results.  

Is it really true that SSC does not handle >1 year weatherfiles correctly?  I ask because I frequently run simulations with weatherfiles that are two years long with 5-minute resolution, and the results have seemed reasonable so far.  My purposes are primarily centered on high-resolution timeseries weather data which is almost never exactly one year in length.  I have been padding the weather files with zeros to fill them out to be an integral number of years long so that SSC will accept them, eg my data runs from June 2017 through August 2018, so I fill in zeros for Jan-June 2017 and Aug-Dec 2018 and run the whole padded weatherfile in a single simulation.  In case it matters, I only use the pvsamv1 module, and in particular, only the `gen` output array -- no financial model or anything.

janinefreeman, 2018-08-24:
@kevinsa5 It is true that SSC does not handle >1 year weatherfiles correctly, however in your _specific_ case, you might not be too far off, if you're processing the `gen` output correctly. 

In pvsamv1, the weather file timestamp is used to calculate sun position, so that is working correctly. However, sunrise/sunset calculations are dependent on timestep, so your 2-year simulations are probably missing generation from both sunrise and sunset hours every day. Fortunately, not a lot of power is produced at those times, so the difference is likely relatively small. The attached PNG shows output from SAM when I take the default Phoenix TMY2 weather file and run it as is (1 yr), versus copying and pasting the entire year to make a 2-yr file and running that. You can see that most of the daylight hours are the same, but sunrise and sunset hours are missing from the 2-yr file. 

[image]

The part where the outputs really start to diverge for multi-year weather files is when SAM calculates _energy_ versus _power_. The `gen` output represents power (kW), so it is independent of timestep length. However, many SAM outputs downstream of that represent energy (kWh), which is dependent on how long SAM thinks the power is produced for. So the annual energy value that SAM would calculate, for example, would be half of what it should be for a 2-yr simulation. Any downstream financial model results would be disastrous, but since you are not using those, you might be ok. If you're processing `gen` correctly, assuming that each power value persists for 1 hour instead of 30 minutes, then you are also probably getting correct energy values.

Hope that helps! We'll be adding better warnings about this case for our next release.

kevinsa5, 2018-08-24:
@janinefreeman this is extremely helpful -- thank you!  The sunrise/sunset phenomenon is good to know about, and I am indeed handling the power-energy conversion correctly (I prefer to work with power specifically for ease of timeseries resampling).  I suppose I got lucky that my misguided multi-year use case can happily ignore any inaccuracy during low-light conditions at dawn and dusk without ill effect. 

Many thanks to you and the team for this tool.  It is the basis for a "mission-critical" tool at my company in utility-scale solar, and we use it for many custom analyses that would be otherwise impossible as well.

dguittet, 2018-09-13:
Modified the check in pv_io_manager to return error specific to stepsPerHour. Also implemented checks in weatherfile reader to check that the stepsPerHour found by subtracting two consecutive records is the same as totalRecords/8760.

commit 70097ac2 (2018-09-13): add subhourly and single-year weather file checks, fixes #192
closed 2018-09-13 by dguittet