NatLabRockies/ssc #196: Improvements for battery modeling in SAM PVWatts model

nickdiorio, 2018-09-27:
_Received from user support_

A few of the operational battery capabilities would be helpful in the PVWatts version. For residential in particular, operational modes to minimize bills on TOU rates, operate to maximize PV self-supply/minimize any export (regardless of rate structure, charge battery when there's excess PV, discharge battery when net load is above 0 until min SOC) would be really helpful. The manual dispatch would be a bonus, but a savvy user can simulate manual dispatch by utilizing the rate structure and a TOU bill minimization mode.

Being able to limit the battery to PV only charging (for ITC) versus allowing grid charging would be really helpful to add to the PVWatts version.

The rate structure tab could house the battery export constraints the same way net metering rules are applied to PV export. Whether or not the battery can export regardless of economics is a critical feature.

The last piece, which is a nice to have but a little outside of scope, would be to enable an 8760 pricing regime for the battery to operate around, simulating wholesale market interactions. That or an 8760 marginal GHG emission signal for the battery to utilize to minimize net GHGs from the system.

frohro, 2020-02-03:
Nick,
I agree that it would be very nice to add a checkbox that would limit the battery to charge only from the PV into PVWatts.  At the moment this is even more important to me because there seems to be a bug in SAM that affects situations where the grid charges batteries.  See [Issue 239](https://github.com/NREL/SAM/issues/239) and [Issue 240](https://github.com/NREL/SAM/issues/240).  I'm doing a feasibility study for Guam, and its rate is flat, so I always only want to use PV to charge the battery.  Charging the battery from the grid and then sending it back to the grid simultaneously unnecessarily cycles the batteries, and loses energy.
Thanks for your work on this!
Rob

brtietz, 2023-12-13:
Closing as "won't fix" - the ITC concerns are no longer relevant, and most of what Nick is discussing seems feasible in PySAM.

closed 2023-12-13 by brtietz