NatLabRockies/ssc #198: Geothermal NCG type is always "JET"

nickdiorio, 2018-10-09:
**Describe the bug**
It appears that there could be multiple options for the NCG type, including: `VAC_PUMP`, `NO_NCG_TYPE`, `HYBRID`, `JET`, but it is currently hard coded to `JET`, within `lib_geothermal.h`.  

**Expected behavior**
When a user selects a power block option with the Powerblock page, it appears they can choose different NCG types, but that information doesn't appear to propagate to SSC.

**Additional context**
This information affects plant cost calculations.

parangat94, 2018-10-17:
As seen in the Excel GETEM model, only the HYBRID NCG type is ever used in calculations. I replaced JET with HYBRID in the code.

closed 2018-10-17 by parangat94

nickdiorio, 2018-10-18:
Thanks @parangat94 !