NatLabRockies/ssc #202: Stack Corruption during NoFinancialModelModuleAndInverterModels test

dguittet, 2018-10-17:
During NoFinancialModelModuleAndInverterModels test in CMPvsamv1PowerIntegration, when module model is set to 2 for CEC User-Entered, there is an error:

> Run-Time Check Failure #2 - Stack around the variable 'm' was corrupted.

'm' here refers to this line:

```
//convert from % to derate
for (size_t m = 0; m < monthlySoiling.size(); m++)
	monthlySoiling[m] = 1.0 - monthlySoiling[m] / 100.0;
```

**To Reproduce**
Run the test in Debug Mode in x64. Does not show up in Win32. Bug is present in both develop and release.

nickdiorio, 2018-10-18:
Thanks for noting @dguittet , my guess is there are still some variables being used before they are allocated or defined.

dguittet, 2018-10-19:
@nickdiorio I've been digging into it and I think it's unrelated to our code... The exception is being thrown during VS debugger's _RTC_CheckStackVars function. It says the stack location that is corrupted is a variable in m:

`module6par m(tech_id, Vmp, Imp, Voc, Isc, beta, alpha, gamma, nser, 298.15);
`

but I've looked and the value there is correctly stored and not modified. Actually, I'm getting warnings that the compiled shared.lib is different from the source code, and I think it's the fact that the debugger is expecting different behavior to be the error.

[image]

Even after cleaning, rebuilding, relinking, I'm still getting this warning... If nobody else has this stack corruption problem, it might just be me. I think it's fine to close this one since SSC logic is ok.