NatLabRockies/ssc #205: Break out Module Losses/Gains Separately

janinefreeman, 2018-11-08:
As we re-do more of the structure of the detailed PV model, it becomes easier to break out module performance losses and gains (temperature, spectral, etc) separately, instead of lumping them all into one "module modeled loss". As shown in the great discussion on #197 , there is a lot of value in breaking these losses out separately and specifying that they are not always losses, but can sometimes be gains.

cristianosaboia, 2020-04-06:
Hi Janine,

I found this discussion after contacting the SAM support regarding the separation of module losses. As I lost the rich discussion on #197, I'll add some comments here.
Regarding temperature and incidence angle losses, I guess everything is settled.
As for other losses, I think they could be divided into spectral and irradiance level losses, which I agree could both be also gains, since they are only a correction in comparison to the STC conditions.

The spectral loss is related to the change in efficiency due to differences on the solar spectrum during the year (in relation to the STC 1.5 AM), and the cell spectral response.

The level of irradiance loss could be treated an independent loss, and is actually described on the single-diode model. If you change the irradiance level on the single diode model, you can see that the cell efficiency is not constant over the irradiance. I saw that SAM already uses the inverse shunt resistance ratio with the irradiance level, as proposed by De Soto et al. 2006, which enhances the single-diode performance on low irradiance levels.

In summary, reading the SAM Photovoltaic Model Technical Reference, it seems to me that at least for the CEC Performance Model, it seems that theses losses are already calculated separately. The spectral loss is calculated with the King et al. (2004), and the level of irradiance loss could be obtained from the difference between the STC efficiency and the efficiency on the real irradiance level, using the single-diode model. So, the breaking out of the losses for these cases would not need major changes. 

Maybe i'm missing out something important or saying what you already know, but I thought it might contribute to the discussion.

janinefreeman, 2023-11-21:
Be sure to include breaking out DC non-linear shading loss separately as well; see discussion in #1090 .

closed 2023-12-13 by janinefreeman