NatLabRockies/ssc #214: SAM tracker angle calculation difference between file types

nickdiorio, 2018-12-14:
Reported from @cdeline 

I am finding some odd behavior in SAM when I try to run the same simulation on a TMY3 and an EPW file.  In the case of the TMY3 file, it seems to run the way I expect – solar position is calculated 30 minutes prior to the timestamp, and GHI_calc matches GHI from the climate file.

However, when I run the same site’s EPW file using DNI&DHI inputs,  the solar position is calculated on the hour, and there’s some big differences between calculated GHI and weather file GHI :

Time stamp | Irradiance GHI SAM calculated \|   (W/m2) | Irradiance GHI from weather file   \| (W/m2)
-- | -- | --
1/2/2018 7:00 | 25 | 25
1/2/2018 8:00 | 225 | 161
1/2/2018 9:00 | 371 | 294
1/2/2018 10:00 | 457 | 395
1/2/2018 11:00 | 479 | 449
1/2/2018 12:00 | 414 | 423
1/2/2018 13:00 | 325 | 362
1/2/2018 14:00 | 222 | 281
1/2/2018 15:00 | 105 | 158
1/2/2018 16:00 | 26 | 26

I suspect that with an EPW file input, there needs to be a -30 minute time shift for calculated solar position inside SAM that isn’t happening.  

I checked earlier versions, and things were working on the half-hour back in 2012 for EPW files, but on the hour by 2014.1.14, so this is an old issue…  There also seems to have been a change between 2018.7.17 and 2018.11.11 because in the July release, the 1-axis tracker orientation was calculated on the half-hour, while in the November release it’s calculated on the hour for EPW files.

Release                EPW GHI calc                      EPW Tracker theta           TMY GHI calc      TMY tracker theta
2018.11.11           T-0                                                          T-0                          T- 0:30                   T-0:30
2018.7.17             T-0                                                          T-0:30                    T-0:30                    T-0:30
2014.1.14             T-0                                                          T-0:30                    T-0:30                    T-0:30
2012.5.11             T-0:30                                                    T-0:30                    T-0:30                    T-0:30

[WeatherFiles.zip](https://github.com/NREL/ssc/files/2680436/WeatherFiles.zip)

dguittet, 2020-09-24:
@janinefreeman We should probably retry this with the new way pvsam is reading the timestamp.

mjprilliman, 2021-10-15:
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/mprillim/AppData/Local/Temp/1/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/mprillim/AppData/Local/Temp/1/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Time   stamp | Irradiance GHI SAM calculated \|   (W/m2) | Irradiance GHI from weather file   \| (W/m2)
-- | -- | --
1/2/2018 7:00 | 25 | 25
1/2/2018 8:00 | 162 | 161
1/2/2018 9:00 | 294 | 294
1/2/2018 10:00 | 396 | 395
1/2/2018 11:00 | 451 | 449
1/2/2018 12:00 | 423 | 423
1/2/2018 13:00 | 363 | 362
1/2/2018 14:00 | 282 | 281
1/2/2018 15:00 | 159 | 158
1/2/2018 16:00 | 26 | 26

</body>

</html>

Looks like the EPW files are reporting the GHI at the top of the hour (minute 0). We were calculating at minute 30 in the current release.

mjprilliman, 2021-10-19:
https://bigladdersoftware.com/epx/docs/8-3/auxiliary-programs/energyplus-weather-file-epw-data-dictionary.html#field-global-horizontal-radiation

janinefreeman, 2023-09-16:
Closed with #626

closed 2023-09-16 by janinefreeman