NatLabRockies/ssc #221: CSP UDPC tables crash when given all same values

Matthew-Boyd, 2019-01-04:
SAM crashes when there is more than one row in a CSP user-defined power cycle (UDPC) table and all the values are the same. This nonsensical input is identified in the code and an exception is thrown, but during the chain of destructor calls an unhandled exception is eventually thrown when an invalid vector is trying to be destroyed.

Vector destructor exceptions seen from:
	c_sco2_cycle_core::S_design_solved::~S_design_solved() in sco2_cycle_templates.h
	c_sco2_cycle_core::S_auto_opt_design_parameters::~S_auto_opt_design_parameters() in sco2_cycle_templates.h
	c_comp_multi_stage::S_des_solved::~S_des_solved() in sco2_cycle_components.h

This problem could stem from the splitting of the single UDPC table into the three component tables when duplicates are removed.

Matthew-Boyd, 2021-10-08:
@tyneises Is this still an issue?

tyneises, 2021-10-11:
@Matthew-Boyd  I think it's fixed in develop; I can't reproduce the issue you describe. Let me know if you can. Otherwise we can close this.
[image]