NatLabRockies/ssc #236: Apply diffuse self-shading in all racking/shade-behavior scenarios

caseyzak24, 2019-02-09:
Currently diffuse self-shading in SAM is applied as shown in the following graph

[image]

where the green scenarios have it applied and the blue scenarios do not.

However, the current diffuse self-shading model, which is based on a simple 2-D unlimited sheds geometry, seems applicable to all scenarios. In particular for backtracking SAT systems, diffuse shading becomes important due to the higher tilt angles experienced and we've seen it contribute up to a 2% loss to energy production.

janinefreeman, 2019-02-11:
Thanks for creating the issue with all the helpful details for this, @caseyzak24 ! The SAM team agrees that diffuse self-shading should be applied in the blue scenarios as well. If you get a chance to make some improvements (or if anyone else is interested in tackling it!), let us know!

caseyzak24, 2019-03-15:
@janinefreeman I am currently working on this issue and I think I have it implemented, just working out a few kinks. I will submit a PR when it is complete.

janinefreeman, 2019-03-19:
Awesome! Looking forward to it! Thanks @caseyzak24 !

janinefreeman, 2019-10-12:
Closed in 27a1793664c34b64b06d06ba055579353f4b55fc

closed 2019-10-12 by janinefreeman