NatLabRockies/ssc #241: Include bifaciality effect in ML module model

timorichert, 2019-02-28:
**Is your feature request related to a problem? Please describe.**
Currently, the ML model does not consider bifaciality effects.

**Describe the solution you'd like**
The backside irradiance should be considered in the effective irradiance sum considering the bifaciality factor, as already implemented in the CEC module model.

caseyzak24, 2019-07-10:
@timorichert this is an important feature for us as well. Are you already working on it? If not I will start but I don't want to duplicate efforts.

caseyzak24, 2019-11-21:
@timorichert just for reference this would be addressed by a commit I submitted as part of PR #314

timorichert, 2019-12-07:
@caseyzak24 great work and thanks for the effort! I'm currently not working on anything SAM-related, but love to see the current developments!

allenlawrence94, 2021-10-05:
Addressed here: https://github.com/NREL/ssc/pull/622