NatLabRockies/ssc #243: Inverter model in pvsamv1 incorrectly using mppt voltage for all hours

caseyzak24, 2019-03-09:
I created this issue as a feature request because I'm not entirely sure I understand the model mechanics...

Right now, in situations where subarray mismatch is not enabled, the total array voltage is calculated as a weighted average of the subarrays. Then array performance is tweaked if it's found that this average voltage lies outside the inverter's mppt bounds. But, this average array mppt voltage (referenced as `PVSystem->p_mpptVoltage`) is still passed to the inverter model. For hours where an mppt boundary voltage was forced on the dc system, shouldn't this also be the voltage that is passed to the inverter performance model?

In the brief time I spent looking at it, it wasn't clear to me what the correct signal is that should be passed into the inverter model, but I'm guessing that it is the weighted average of the subarray voltage after the mppt window is applied.

If someone can point in the right direction of what signal should be applied I can implement the fix.

caseyzak24, 2019-03-09:
On second thought, perhaps the easiest fix is just to check the mppt window again right before assignment to the inverter input voltage, and set the voltage to the mppt boundary value if needed.

This wouldn't result in any sort of output that the user could see (something like inverter1_DCVoltage to go alongside inverterMPPT1_DCVoltage) but it would result in what I believe is the correct behavior.

If someone can confirm that this is actually an issue that needs fixing I can take care of it sometime in the near future.

janinefreeman, 2019-03-11:
Hi Casey, thanks for bringing this up. I can take a look and see what's going on. I suspect that it may be related to the fact that we can't re-calculate module performance for some module models (Sandia and single point efficiency models), so we may need to pass in what the voltage "would have been" to get the correct amount of clipping, but I'm not certain, so I'll investigate!

janinefreeman, 2019-05-08:
@caseyzak24 Hi! Finally got to take a look at this, and I think you may be right, p_mpptVoltage needs to get assigned after the recalculation. I'll take a look at integrating that, since I want to also clean up those variables so that they're easier to follow. Thanks for digging into this!

commit 4f92e283 (2019-12-11): updated pvsamv1 multiple mppt model to correctly re-assign voltage going into inverter if inverter changes module voltages due to MPPT bounds. Added a test for a related item for a bug I introduced fixing this bug. Closes #243
closed 2019-12-11 by janinefreeman