NatLabRockies/ssc #247: Battery calendar life model may have minor bug

nickdiorio, 2019-04-03:
As reported by @cdeline, there’s a term k_cal that has a dependence of `(SOC / T) – (1/T)` .  Based on the literature, it appears this should perhaps instead be `(V_batt / T) – (V_batt_0 / T)` where `V_batt_0 = 3.6V`.  Therefore, at 100% SOC, you have a bit of a higher terminal voltage like 4.1V which accelerates this calendar degradation loss.  

Need to investigate this and confer with transportation researchers.

janinefreeman, 2021-08-03:
Is this possibly addressed in your new lifetime model, @dguittet ? Or is this separate?

dguittet, 2021-08-05:
No, this is an issue with the original calendar model. The new lifetime models have entirely separate equations.

closed 2024-09-25 by janinefreeman