NatLabRockies/ssc #252: Coersion to ssc_number_t results in loss of precision

nickdiorio, 2019-04-16:
**Describe the bug**
It appears that when doubles are converted to `var_data` and coersed to `ssc_number_t`, there are differences in the original and output calculations.  A good example of this is to consider a very large system, and then look at the annual energy calculated within the compute module, and then the value that is returned to the user-interface.

dguittet, 2019-04-16:
ssc_number_t is a float. We should consider moving to doubles to maintain the same precision as we use in calculations.

nickdiorio, 2019-04-16:
Good point, I thought we were dynamically deciding whether `ssc_number_t` was a double or float based on the OS.  Unclear that we need to maintain as floats in today's computational age.

nickdiorio, 2019-04-23:
We've moved `ssc_number_t` to `double`, in the `develop` branch.

closed 2019-04-23 by nickdiorio