NatLabRockies/ssc #277: Bifacial rear irradiance reporting error - rear POA is multiplied by bifaciality coeff.

cdeline, 2019-06-17:
Hi everyone.  I'd like to report an error in how SAM is reporting rear irradiance which might cause some confusion.  This doesn't cause any errors in the final output of system power, but intermediate parameters are being incorrectly reported.

At the moment, it appears that the value returned in the SAM Data Tables 'Subarray 1 POA rear total irradiance after reflection (IAM) (W/m2)'  is reported as the rear irradiance value *times the bifaciality coefficient*.  This will cause significant under-reporting of the rear irradiance, for the case of people trying to compare measured back POA irradiance with modeled expected values.  

It does appear that the 'Subarray 1 POA total irradiance' is being calculated more or less correctly by adding POA_front + POA_rear * bifaciality coeff.  However, please change the intermediate value reported for POA_rear to not include this bifaciality coefficient.  THANKS!
This may be related to #191, not sure.

-Chris

nickdiorio, 2019-08-06:
Possibly we need to update the inputs to the module model to capture this change, which no longer includes bifaciality in poaRear: 

[image]

nickdiorio, 2019-08-06:
@caseyzak24

commit ab65852b (2020-02-13): updated bifacial model so that bifaciality is applied to rear irradiance in module model. bug was introduced in 5c2a130 and is present in 2020 Beta version but no other released versions. fixes #277 fixes #334
closed 2020-02-13 by janinefreeman