NatLabRockies/ssc #286: Allow an option to enable/disable battery charge from grid/PV in battwatts

gauravumrani, 2019-07-02:
## Description
I am using `battwatts` module, I do not want the battery to get charged from the grid
basically `grid_to_batt` **(Electricity to battery from grid)** in `battwatts` should return empty array/response

In this file, it is set to true always, if we can set this to true/false from the API, it would be great
https://github.com/NREL/ssc/blob/develop/ssc/cmod_battwatts.cpp#L228

gauravumrani, 2019-07-03:
@nickdiorio ?

janinefreeman, 2019-07-03:
The battwatts module is a simplified set of battery modeling assumptions, chosen to allow only a few common pathways in order to provide the least confusion (but also the least flexibility) to the user. However, disabling grid charging (and a lot of other functionality!) is available if you use the detailed PV model with the battery model!

closed 2019-07-03 by janinefreeman