NatLabRockies/ssc #305: Not able to build SSC library in Ubuntu

gauravumrani, 2019-08-13:
Hey guys, I am trying to build SSC code in my ubuntu system.
in `path/sam_dev/ssc/build_linux, when I enter `make` command, it does not work.
It is working when I build ssc for macos, `make` command in  `path/ssc/build_osx` from my macbook.
```
GCC version : 7.4.0
G++ version: 7.4.0
LDD version:  2.27
```
Please help, I just want to build ssc library
I am getting following errors in ubuntu.

```
In file included from ../splinter/Core:263:0,
                 from ../splinter/definitions.h:36,
                 from ../splinter/bsplinebasis.h:13,
                 from ../splinter/bsplinebasis.cpp:10:
../splinter/src/Core/arch/SSE/PacketMath.h:29:39: error: ignoring attributes on template argument ‘__m128 {aka __vector(4) float}’ [-Werror=ignored-attributes]
 template<> struct is_arithmetic<__m128>  { enum { value = true }; };
                                       ^
../splinter/src/Core/arch/SSE/PacketMath.h:30:40: error: ignoring attributes on template argument ‘__m128i {aka __vector(2) long long int}’ [-Werror=ignored-attributes]
 template<> struct is_arithmetic<__m128i> { enum { value = true }; };
                                        ^
../splinter/src/Core/arch/SSE/PacketMath.h:31:40: error: ignoring attributes on template argument ‘__m128d {aka __vector(2) double}’ [-Werror=ignored-attributes]
 template<> struct is_arithmetic<__m128d> { enum { value = true }; };
                                        ^
../splinter/src/Core/arch/SSE/PacketMath.h:101:43: error: ignoring attributes on template argument ‘Eigen::internal::Packet4f {aka __vector(4) float}’ [-Werror=ignored-attributes]
 template<> struct unpacket_traits<Packet4f> { typedef float  type; enum {size=4}; };
                                           ^
../splinter/src/Core/arch/SSE/PacketMath.h:102:43: error: ignoring attributes on template argument ‘Eigen::internal::Packet2d {aka __vector(2) double}’ [-Werror=ignored-attributes]
 template<> struct unpacket_traits<Packet2d> { typedef double type; enum {size=2}; };
                                           ^
../splinter/src/Core/arch/SSE/PacketMath.h:103:43: error: ignoring attributes on template argument ‘Eigen::internal::Packet4i {aka __vector(2) long long int}’ [-Werror=ignored-attributes]
 template<> struct unpacket_traits<Packet4i> { typedef int    type; enum {size=4}; };
.
.
.
.
.
_some more errors_
.
.
.
.
../splinter/bsplinebuilder.cpp:558:1:   required from here
../splinter/src/Core/DenseCoeffsBase.h:56:30: error: ignoring attributes on template argument ‘Eigen::internal::packet_traits<double>::type {aka __vector(2) double}’ [-Werror=ignored-attributes]
                      >::type PacketReturnType;

../splinter/bsplinebuilder.cpp:558:1:   required from here
../splinter/src/Core/Assign.h:53:57: error: enum constant in boolean context [-Werror=int-in-bool-context]
     MayLinearVectorize = MightVectorize && MayLinearize && DstHasDirectAccess
                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
../splinter/src/Core/Assign.h:57:41: error: enum constant in boolean context [-Werror=int-in-bool-context]
     MaySliceVectorize  = MightVectorize && DstHasDirectAccess
                          ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
<builtin>: recipe for target 'bspline.o' failed
make[1]: *** [bspline.o] Error 1
cc1plus: all warnings being treated as errors
<builtin>: recipe for target 'bsplinebuilder.o' failed
make[1]: *** [bsplinebuilder.o] Error 1
make[1]: Leaving directory '/path/sam_dev/ssc/build_linux'
Makefile:2: recipe for target 'all'

```

gauravumrani, 2019-08-16:
I tried building it with gcc version 4.8.5, it works but ssc.so file becomes too large (138 mb)

gauravumrani, 2019-08-16:
@janinefreeman ?

dguittet, 2019-08-16:
Regarding the first error you had, all the makefiles have `-Werror` enabled. For instance, in Makefile-splinter: 

```
WARNINGS = -Wall -Werror -Wno-strict-aliasing -Wno-deprecated-declarations -Wno-unknown-pragmas -Wno-reorder
```

The warnings change with compiler version. You can remove -Werror from Makefile-splinter and from any other makefile of a project what produces some type of "-Werror=" message.

janinefreeman, 2019-10-12:
@sjanzou , can you share your build process on Ubuntu?

dguittet, 2020-09-23:
@gauravumrani Is this still an issue you're dealing with, or can I close this?

gauravumrani, 2020-09-23:
you can cloy this now, thanks

closed 2020-09-23 by dguittet