NatLabRockies/ssc #347: Excessive charging from the grid during custom dispatch

brtietz, 2020-03-26:
**Describe the bug**
When a battery with a custom dispatch profile charges from the grid, an order of magnitude more energy is being recorded under "Electricity to battery from grid" than is actually reaching the battery. As a result, the energy from the grid is much larger than the energy dispatched.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a Detailed PV model with battery storage and a single owner model (we have not reproduced this on residential) 
2. Select custom dispatch and paste in the attached values: 
[custom_dispatch_singleowner_hourly.csv.zip](https://github.com/NREL/ssc/files/4389611/custom_dispatch_singleowner_hourly.csv.zip)
3. Allow charging from the battery and simulate
4. Check in the data tables "batt_annual_charge_energy" and "batt_annual_discharge_energy" the charge energy will be unexpectedly large due to grid charging. Additionally, the power provided to the battery by the grid is larger than the adjustments in the battery's SOC.

**Expected behavior**
The above values should be within the expected charging losses. Battery charging from the grid's values should match the change in state of charge more closely.

**Desktop (please complete the following information):**
 - Version 2020.2.29, develop branch

brtietz, 2020-03-26:
Bug also exists in patch branch.

brtietz, 2020-03-26:
Unit test capturing the issue is available on branch excessive_grid_charging_347. A key variable seems to be a DC connected battery, this can be produced in a residential (behind the meter) system as well.

dguittet, 2020-03-29:
issue fixed in #350 

tests updated in pv battery integration test branch

PR #350 (2020-03-29): update battery model to fix FOM dispatch bugs
- fix battery_t::calculate_max_charge_w & calculate_max_discharge_w
- fix check_constraints for DC_CONNECTED systems where power flow to battery was miscalculated
- increase tolerance for error checking in check_constraints
- fix SharedInverter copy constructor
- rewrite SharedInverter::calculateRequiredDCPower to be more explicit about no persistent state changes
- fix enforcement of same-sign AC power as DC power for SharedInverter::calculateACPower
fixes #347

- remove index updating in FOM automatic dispatch, which was skipping steps in subhour [... cut]
closed 2020-03-29 by dguittet