NatLabRockies/ssc #364: Warning for tilted, tracking, bifacial systems

janinefreeman, 2020-05-08:
The bifacialvf code that we use to calculate bifacial gain in SAM may not be suited for trackers with non-zero tilt. Would like to add an additional output warning to SSC if a user tries to model this combo.

mjprilliman, 2020-09-21:
I believe this issue has already been addressed in a previous release. Unless there is further explanation that is needed in this warning this issue can be closed. 

[image]

cpaulgilman, 2020-09-25:
This requires an additional warning for when a bifacial module is used with tracking: Bifacial model is based on fixed arrays.

cpaulgilman, 2021-01-06:
The message appears for bifacial modules with one-axis tracking regardless of the tilt angle. It should only appear when the tilt angle is not zero: https://sam.nrel.gov/forum/forum-general/3285

cpaulgilman, 2021-01-28:
[image]

[image]

commit f91f7419 (2020-09-25): Fix #364 Add warning when bifacial is used with tracking for pvsam and pvwattsv7

Also clean up all log() message in pvsam and pvwattsv1: Capitalize first word and revise for clarity.
commit 728c55c9 (2020-09-25): Merge pull request #455 from NREL/SSC364_fix_pv_warnings

Fix #364 Add warning when bifacial is used with tracking for pvsam an…
commit 9aafbf2d (2021-01-28): Fix #364 error messaging for PV one-axis tracking and bifacial
commit dfcaf1bc (2021-01-29): Revert "Fix #364 error messaging for PV one-axis tracking and bifacial"

This reverts commit 9aafbf2da5ee0ff762605ddb02d63bff6a109ebd.
closed 2021-01-28 by cpaulgilman