NatLabRockies/ssc #368: ssc installation error

CLLEROY, 2020-05-14:
Hello,

Following instructions here:
https://github.com/NREL/SAM/wiki/Linux-Build-Instructions
At step 5, installing ssc is not working, you'll find below the errors encountered (*)?
Any Idea to solve this?
Thanks in advance
Christine
(*)
[100%] Building CXX object ssc/CMakeFiles/ssc.dir/vartab.cpp.o
Reaping winning child 0x226fa30 PID 187880 
Live child 0x226fa30 (ssc/CMakeFiles/ssc.dir/vartab.cpp.o) PID 187881 
sam_dev/ssc/ssc/vartab.cpp: In copy constructor ‘var_table::var_table(const var_table&)’:
ssc/ssc/vartab.cpp:193:56: error: no matching function for call to ‘var_table::var_table()’
 var_table::var_table(const var_table &rhs) : var_table() {
                                                        ^
/home/prog/SAM_LITEN/sam_dev/ssc/ssc/vartab.cpp:193:56: note: candidate is:
In file included from /home/prog/SAM_LITEN/sam_dev/ssc/ssc/vartab.cpp:24:0:
/home/prog/SAM_LITEN/sam_dev/ssc/ssc/vartab.h:48:5: note: var_table::var_table(const var_table&)
     var_table(const var_table &rhs);
     ^
sam_dev/ssc/ssc/vartab.h:48:5: note:   candidate expects 1 argument, 0 provided
Reaping losing child 0x226fa30 PID 187881 
make[2]: *** [ssc/CMakeFiles/ssc.dir/vartab.cpp.o] Error 1
Removing child 0x226fa30 PID 187881 from chain.
Reaping losing child 0x191ebb0 PID 187334 
make[1]: *** [ssc/CMakeFiles/ssc.dir/all] Error 2
Removing child 0x191ebb0 PID 187334 from chain.
Reaping losing child 0x1bac7a0 PID 186056 
make: *** [all] Error 2
Removing child 0x1bac7a0 PID 186056 from chain.
sam_dev/ssc/build>>

dguittet, 2020-05-27:
Sorry for the delayed response. I fixed something in `vartab.h` last week for linux build issues: https://github.com/NREL/ssc/commit/5fb1c44f87be3addcf050f39dff08cad300bc3b0

Do you mind trying to build again?

If it doesn't work, could you share details about your platform?

CLLEROY, 2020-06-05:
ssc seems to be built correctly now, but maybe not, as I get this error for SAM now:

```
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
SSC_LIB
    linked by target "SAM" in directory /home/prog/SAM_LITEN/sam_dev/sam

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
>> 
```

I have only this lib:
```
ls $SSC_LIB
CMakeFiles  cmake_install.cmake  dltest  Makefile  **ssc.so**

```

dguittet, 2020-06-05:
It looks like you're building each repo separately. Building with an [CMakeLists.txt](https://github.com/NREL/SAM/wiki/Linux-Build-Instructions#5-install-lk-wex-ssc-then-sam) at the `sam_dev` folder is the easiest as it'll build lk, wex, ssc & sam altogether.

Otherwise, for the separate build, the projects need to be built in the correct folder, i.e. `sam_dev/ssc/build/ssc/ssc.so` with $SSCDIR=.../sam_dev/ssc.

We have an upcoming change #363 that will allow you to point to the library files with env variables the way you are trying, but for now, the `SSC_LIB` referred to in your error message is actually a CMake variable.

dguittet, 2020-06-06:
I just pushed in changes that will allow you to use environment variable inputs to CMake, where `SSC_LIB`, `WEX_LIB` and `LK_LIB` are env variables to the folders containing those libraries. For the debug build, use `SSCD_LIB`, etc.

CLLEROY, 2020-06-17:
Many thanks, it seems ok:
[100%] Built target SAM_api

closed 2020-06-17 by CLLEROY