NatLabRockies/ssc #379: Ancillary losses batt

dguittet, 2020-06-01:
The ancillary losses in the battery model do not have any effect on the final power. This is due to the `losses_t` class being unused.

The losses should probably be applied before the dispatch decision, so that if the target is 10 kW and the losses is 1kW, the dispatch algorithms will meet the target 10kW but the battery will have discharged 11kW.

cpaulgilman, 2020-06-17:
See https://sam.nrel.gov/forum/forum-general/3018 for a user example file for Lithium-ion (LFP/Graphite) and vanadium flow battery.

janinefreeman, 2020-08-05:
@dguittet was this issue resolved, or is this still on our to-do list?

brtietz, 2020-08-07:
A sample SAM file subject to this issue is attached:
[battery-ancillary-losses-issue.zip](https://github.com/NREL/ssc/files/5043607/battery-ancillary-losses-issue.zip)

brtietz, 2020-10-02:
@dguittet I'm digging into this and I've hit a scope question. I've added the following line to the dispatch classes to make sure the losses are applied to the power flow:

`m_batteryPower->powerSystemLoss = _Battery->get_state().losses->loss_percent;`

There's a naming issue there, I'm thinking about changing `loss_percent` to `loss_kw`, but that's a side issue for now.

Right now, P_system_loss_ac in BatteryPowerFlow is only applied to P_gen_ac. This seems insufficient, I think these ancillary losses would affect a battery's ability to meet load in a behind the meter system as well. Is it a good idea to attempt to apply these losses to more places in BatteryPowerFlow, or should I assume BatteryPowerFlow is correct and only work in the dispatch classes?

Happy to jump on Teams to discuss if that's easier.

brtietz, 2020-10-02:
A couple of decisions made this afternoon:
- The current losses model will not change the current within battery_t. Dispatch and powerflow code need to account for the losses when specifying current or power to the battery.
- Losses that do change the current within the battery will be handled after the release in issue 456
- By default, charging and discharging losses will be met by the battery. In situations where the battery is unable to meet these losses, the grid can step in. (the most probable situation here is an erroneous dispatch)
- Idle losses will be handled by PV or the grid.
- Losses need to be on the same side of the inverter as the battery (powerflow currently only has a system_loss_ac variable, this needs to be amended)