NatLabRockies/ssc #383: Generalize battery power flow outputs

dguittet, 2020-06-02:
In the GUI, there's no standalone battery or pairing of batteries with another technology besides PV. In the SDK & PySAM, battery + wind or just battery is possible. The outputs of the battery compute module such as `pv_to_grid`, `pv_to_batt` should be changed to more general terms such as `generator_to_grid`, `generator_to_batt`.

cpaulgilman, 2020-06-02:
In 7bcb572a02f1b3d3199431a5701eb20d34e6ca84, I changed the variable labels to refer to 'system' instead of 'pv' but did not change the variable names themselves. I'm not sure what name to use, "generator" might suggest a fuel-powered genset, but "system" might appear to include the battery.

dguittet, 2020-06-03:
Thanks @cpaulgilman glad to know one half of this is done already.

I am also not sure what's the best nomenclature... I am remembering unresolved naming issues around `gen`. In particular how it feeds into the financial models and whether the power going to the `load` was expected or not depends on if it was BTM or FOM-- from the bug @sjanzou resolved. I think we discussed a separate output from `gen` for the case where the power going to the load has been subtracted called `system_to_grid`.

How about `plant_to_grid`? We could also discuss at the meeting.

dguittet, 2020-07-07:
Working with @brtietz on generic-battery, we noticed it's very confusing that "System Power Generated" is both the power plant + battery since there isn't a separate output for just the power plant/pv.

I think we need to either make a new output for just the power plant/pv or a new one for when battery is included. @cpaulgilman thoughts on naming?

cpaulgilman, 2020-07-14:
I like gen_with_storage.

cpaulgilman, 2020-09-22:
https://github.com/NREL/ssc/commit/ce96913e5c228bb96b702211269a8d0c8efd571d fixes SAM383 not SSC 383.

brtietz, 2020-10-13:
@dguittet - should I take this issue to solve along with https://github.com/NREL/SAM/issues/370? `gen_with_storage` is a good variable name for the "net energy delivered to the grid" variable we discussed at last week's meeting.

dguittet, 2020-10-14:
@brtietz Yes, that would be awesome!

brtietz, 2020-11-11:
Closed with https://github.com/NREL/ssc/commit/e96e00f476f0af3dcefe8c5f78d3157fabe66690

commit ce96913e (2020-09-22): Fix #383 DC power labels
closed 2020-11-11 by brtietz