NatLabRockies/ssc #400: Transformer no load loss units?

janinefreeman, 2020-06-12:
It apppears that the "xfmr_nll_ts" output may in fact be in kWh rather than kW as it's labeled:

from lib_pv_io_manager.cpp line 805:
`p_transformerNoLoadLoss = cm->allocate("xfmr_nll_ts", numberOfWeatherFileRecords);`

from cmod_pvsamv1:
line 2154: `ssc_number_t xfmr_nll = PVSystem->transformerNoLoadLossFraction * static_cast<ssc_number_t>(ts_hour * transformerRatingkW);`
line 2164: `ssc_number_t xfmr_loss = xfmr_ll + xfmr_nll; // kWh`
line 2182: `PVSystem->p_transformerNoLoadLoss[idx] = xfmr_nll;`

@sjanzou could you please confirm or contradict when you have a chance?

sjanzou, 2020-06-16:
This issue was first reviewed and closed in pull request 353 and SAM issue 274. The timestep values were indeed in kWh and have been updated to kW and tested with project files from SAM issue 274 and updated in ssc_400 branch. See pull request #401

sjanzou, 2020-06-16:
Verification of 30 min weather file and annual values and comparison with hourly data from SAM  274 and pull request #353  
[GitHubIssuesSAM_274_ssc_400.docx](https://github.com/NREL/ssc/files/4785437/GitHubIssuesSAM_274_ssc_400.docx)
[30min sub hourly.xlsx](https://github.com/NREL/ssc/files/4785441/30min.sub.hourly.xlsx)

sjanzou, 2020-06-21:
Closed with pull request #401

closed 2020-06-21 by sjanzou