NatLabRockies/ssc #404: Vanadium flow model discharges more than it charges

dguittet, 2020-06-17:


dguittet, 2020-06-18:
According to the [paper](https://ieeexplore.ieee.org/document/6971196) representing the implemented model, the open-circuit voltage and the voltage drop forced by the aggregated resistance (R_specific) of the battery stack is:

[image]

where the convention is that voltage and current are both negative during charging. So during charging, `V_stack < 0`, `I_stack < 0` and `R_specific > 0`, and the two terms accumulate in absolute value.

Current SAM code has the two terms with opposite directions when charging since the voltages always > 0 but current < 0 when charging:
```
_V_ref_50 + m_RCF * T * A + I_string * _R
```

This resulted in a different voltage versus SOC curve for charging and discharging:
[image]

This resulted in a higher dE/dSOC magnitude for discharging than charging, and therefore led to incorrectly high battery efficiencies.

The original `voltage_vanadium_redox_t` code shows a negative for I * R: https://github.com/NREL/ssc/blob/2ed1ddbaf684346bf7ba8577c7e043a114b4216e/shared/lib_battery.cpp#L730

This seems like it could create the opposite effect and result in lower battery efficiencies. 

However I think this voltage model only intends to calculate the battery system's OC voltage and does not model overpotential. In which case, the OC V versus SOC curves should be the [same](Electrochemical) whether charging or discharging:
[image]

This is consistent with the voltage model for other chemistries which also assumes the same characteristics for the charge and the discharge cycles.

So I'd propose using the same voltage characteristics:
```
_V_ref_50 + m_RCF * T * A + fabs(I_string) * _R
```

cpaulgilman, 2020-06-18:
This comment on the forum thread suggests there may be an issue for Li-ion and other chemistries as well?

https://sam.nrel.gov/forum/forum-general/3018-battery-efficiencies.html?start=6

brtietz, 2020-06-18:
@dguittet - yes, that seems like a reasonable way to match the sign conventions in the paper. The paper also allows the resistance to change between charging and discharging, is that something we need to include?

@cpaulgilman Can we get the .sam file for that new case? https://sam.nrel.gov/forum/forum-general/3009 suggests this is not an issue for the default Li-ion parameters, so it'd be good to know which parameters are causing the issue.

dguittet, 2020-06-19:
@brtietz Thanks for the input! We have a task to add changing resistance with temperature, but not with charging or discharging. I think it's interesting to include! It may help with modeling some of the difference in V vs SOC in other use cases besides VRF as well. We'd probably need to first think about those use cases and what the values may be.

@cpaulgilman I forgot to mention that I had to "turn off" the thermal model by turning up the heat transfer coefficient (bug fix upcoming in next release-- maybe patch at this point?) before I was able to see the voltage issue. The thermal model outputs were sometimes affecting charging disproportionately:

[image]

dguittet, 2020-06-19:
I looked more into the other voltage model, the Shephard-Tremblay one, and that one seems to take into account polarization and produces higher voltages during charging.

Then with regards to different resistances during charge or discharge, it seems only applicable to vanadium flow batteries.

cpaulgilman, 2020-06-19:
@brtietz Here's a link to a .sam file with a Li-ion case that shows the same issue as the vanadium flow case:

https://sam.nrel.gov/media/kunena/attachments/101439/VRF_2020-06-19.sam

And, for reference the forum post where it is posted:

https://sam.nrel.gov/forum/forum-general/3018-battery-efficiencies.html?start=6#8329

Thanks,
Paul.

brtietz, 2020-06-22:
@cpaulgilman This looks like a general problem with the voltage table. Since the voltage table is only a function of DOD, the battery will spend some of its time charging at a lower voltage than it discharges.

@dguittet My intuition on this is to add a `-I * R` term to the voltage table's calculate_voltage_for_current function, to account for the cyclic losses similar to the Shephard-Tremblay electrochemical model. Thoughts?

commit cd1dbb01 (2020-06-19): vanadium voltage model use same curve for charge and discharge; fix #404

add test with higher power to show failure before and success now
commit 1b0e20b7 (2020-06-22): Vanadium flow battery: voltage model fixes (#407)

* copy over vanadium flow code fixes from develop; add test for roundtrip eff

* vanadium voltage model use same curve for charge and discharge; fix #404

add test with higher power to show failure before and success now

* fix vanadium voltage::calculate_max_charge_w  to return current of whole stack

* comment out test print statement

* add comment about why using fabs(I_string)
closed 2020-07-02 by dguittet