NatLabRockies/ssc #412: Voltage table may result in roundtrip eff > 100

dguittet, 2020-06-22:
@brtietz Here's a link to a .sam file with a Li-ion case that shows the same issue as the vanadium flow case:

https://sam.nrel.gov/media/kunena/attachments/101439/VRF_2020-06-19.sam

And, for reference the forum post where it is posted:

https://sam.nrel.gov/forum/forum-general/3018-battery-efficiencies.html?start=6#8329

Thanks,
Paul.

_Originally posted by @cpaulgilman in https://github.com/NREL/ssc/issues/404#issuecomment-646855931_

---------

@cpaulgilman This looks like a general problem with the voltage table. Since the voltage table is only a function of DOD, the battery will spend some of its time charging at a lower voltage than it discharges.

@dguittet My intuition on this is to add a `-I * R` term to the voltage table's calculate_voltage_for_current function, to account for the cyclic losses similar to the Shephard-Tremblay electrochemical model. Thoughts?

_Originally posted by @brtietz in https://github.com/NREL/ssc/issues/404#issuecomment-647606082_

dguittet, 2020-06-24:
@brtietz I wonder if it makes sense to add an external term since I think of the voltage table model as incorporating everything that affects voltage-- polarization, internal or aggregated resistance, etc. So if I had the curve output from the vanadium model for instance and put that into the table, would it make sense to add another IR loss to it?

I think hysteresis was only explicitly added for Lead acid, NiMH, and NiCD in newer [versions](https://pdfs.semanticscholar.org/8f16/68ffef08c83a3a69f8f3c557d04cd9a5ffc2.pdf) of the Shephard Tremblay model. 

I wonder how much of the cyclic losses we see in SAM is an artifact of how the integral P = VI is calculated as a single right-hand Riemann sum using voltage at the end of the time step. With the actual integral, the roundtrip efficiency should always been 100% for the voltage table.

I remember you fitted the kokam data to the shephard tremblay parameters. Could you please run a few a cycles comparing SOC vs Voltage during charging and discharging using just `_E0 - _K * (Q_cell / (Q_cell - it)) + _A * exp(-_B0 * it) - R * I` and see how the cyclic losses compare?

brtietz, 2020-06-24:
Can you expand on what you're thinking as far as an external term? The voltage table as currently implemented can't model hysteresis, we would need to be able to specify different curves for charging and discharging. I'm open to that approach, but feel as though it's a larger burden on the user than simply specifying the internal resistance.

I can do that comparison, but it'll be Friday at the earliest.

dguittet, 2020-06-24:
I agree it would be good to have some way to represent charging vs discharging curves.

I actually meant that I think of the suggested `IR` term as external to the voltage curve that's being represented by the voltage table model. In my mind the original curve could be the open circuit voltage, or it could be discharge cell voltage with overvoltage factored in already, or it could be the voltage at the pack level. In the first case, adding a resistance term that increases the charge voltage and decreases the discharge voltage would make sense and be helpful.

The other confusing part is that the resistance term is currently the same one in the voltage model and in the thermal model, but is used differently in the vanadium model where it's at the aggregate, stack-level and in the proposed case here, it's at the cell level.

I don't think any of that disqualifies adding `IR`, which sounds reasonable to me, but the right-hand Riemann sum strikes me as a potentially bigger issue when it comes to just round trip efficiency at the end of the day-- but I haven't had time to dig into that yet!

brtietz, 2020-06-29:
A quick run of the Kokam voltage model in Python, seems to suggest a left-handed (voltage calculated using the depth of discharge at the start of the step) sum will charge due to the numerical integration, whereas a right handed sum will discharge.

SOC % at 3 cycles of constant power (52.8 watts), left handed sum:
> 0.20328701575794628
0.585421874728884
0.20340324594872217
0.5855341586828566
0.20351950322756052
0.5856464678700413

SOC % at 3 cycles of constant power (52.8 watts), right handed sum (voltage based on DOD at end of step):
> 0.20328623995018924
0.5853080707146636
0.20317003622672658
0.5851958373911357
0.20305385951617183
0.5850836291976065

With R = 0.0015 Ohms, and I = 15 Amps the -R * I term barely shows up in the voltage, I don't see it until the 15th significant figure.

The code that generated the above is in the SAM-analysis repository if you want to check it or test something else.

brtietz, 2020-06-30:
Thinking about these results some more, I'm surprised `_Battery->run` takes current as an input instead of power. If we passed in power instead, we could include voltage in the iterative loop, and approximate midpoint or trapezoidal numerical integration for more accurate results.

dguittet, 2020-07-06:
Thanks for running the sanity checks but I'm afraid I don't understand the comparison you're doing with right and left handed integration.

Given how the SAM code works currently with right-handed integration, what is the error between calculated SOC and power/V/I versus the actual data?

> Thinking about these results some more, I'm surprised `_Battery->run` takes current as an input instead of power. If we passed in power instead, we could include voltage in the iterative loop, and approximate midpoint or trapezoidal numerical integration for more accurate results.

Yeah I agree that `run` would make more sense using power as input, but current is the input for the thermal & capacity models so that may be why? 

For computing power, I personally would first try to derive an analytic integration function and use a numerical solver to find the required current, at least for SSC C++ code. Not sure about the trade-offs between adding a numerical integration alg versus relative numerical stability / convergence of each approach though. 

I'd also propose seeing if using the 2009 version of the Shephard-Tremblay model helps increase accuracy.

dguittet, 2020-07-06:
Not sure what you or @cpaulgilman think, but I'd be fine saying the voltage table is a rough model that doesn't capture hysteresis, different charge vs discharge behaviors, etc and if a user would like those, to use the voltage model; otherwise, they could modify the battery losses (once those are fixed).

brtietz, 2020-07-06:
I took a second look at this, and it looks more like an accounting issue when trying to limit the power for a system that can't grid charge. Notice how in these early steps (11 - 17), the charging current is the same, but the variation in power does not correspond to the variation in voltage:

```
-9	-125.765	948.601	11.3358	225.015
-31	-125.765	1008.64	17.6715	350.78
0	0	1008.64	17.6715	350.78
0	0	1008.64	17.6715	350.78
0	0	1008.64	17.6715	350.78
-59	-125.765	1026.76	24.0073	476.545
0	0	1026.76	24.0073	476.545
```

This line (275 - 282 in patch) in lib_battery_powerflow is suspicious, since this can be called after all updates to current and SOC:

```
	// Error checking trying to charge from grid when not allowed
	if (!m_BatteryPower->canGridCharge && P_battery_ac < -tolerance){
	    if ((fabs(P_grid_ac - P_grid_to_load_ac) > tolerance) && (-P_grid_ac > P_grid_to_load_ac)) {
            P_battery_ac = P_pv_ac - P_pv_to_grid_ac - P_pv_to_load_ac;
            m_BatteryPower->powerBatteryDC = -P_battery_ac * m_BatteryPower->singlePointEfficiencyACToDC;
            return calculateACConnected();
        }
	}
```

I think it'll be easier to explain the numerical integration tests in a meeting, but I no longer think that's our core problem. I'd be willing to accept results from the voltage table with 100% efficiency, but we need to make sure charging power is less than or equal to discharged power before closing this issue.

dguittet, 2020-07-06:
Is that from https://sam.nrel.gov/media/kunena/attachments/101439/VRF_2020-06-19.sam? That table doesn't have column headers :P 

It's very possible that piece of code is messing up the powerflow calculations as it was added to patch up non-convergence issues in the constraints checker.

brtietz, 2020-07-06:
Yes, though I had to look at current and voltage explicitly. Rather than trying to edit that comment, here's a screenshot. 
[image]

Also worth noting, power to load + power to battery > power from system

closed 2020-07-15 by brtietz