NatLabRockies/ssc #420: Update PVWatts module type options

cpaulgilman, 2020-07-02:
For the crystalline silicon modules, I suggest updating your options for quality/efficiency of pv module. Currently, the help system identifies the options as:

PVWatts® Module Type | Cell Material | Approximate Nominal Efficiency | Module Cover | Temperature Coefficient of Power
-- | -- | -- | -- | --
Standard | Crystalline Silicon | 15% | Glass | -0.47 %/°C
Premium | Crystalline Silicon | 19% | Glass with anti-reflective coating | -0.35 %/°C

Consider updating this table to have Standard and Premium modules that are older and newer. Something like the following:

PVWatts® Module Type | Cell Material | Approximate Nominal Efficiency | Module Cover | Temperature Coefficient of Power
-- | -- | -- | -- | --
2015 Standard | Crystalline Silicon | 14% | Glass | -0.47 %/°C
2020 Standard | 16% | Glass with anti-reflective coating | -0.37 %/°C
2015 Premium | 17% | Glass with anti-reflective coating | -0.32 %/°C
2020 Premium | 20% | Glass with anti-reflective coating | -0.28 %/°C

Or add in an option to enter Beta Voc (and NOCT/NMOT) as part of System Information specs.

cpaulgilman, 2020-07-02:
> CdTe technology and the CIGS technology in efficiency, and they ae both lumped together as “thin film”. And likewise, there is some separation in the monocrystalline PV market between heterojunction vs other device structures in the temperature coefficients.
>
> My request is to add a 4th module type called “custom”, and if that option is specified then there are two new inputs for module efficiency at STC and temperature coefficient of efficiency. I believe that’s all that you need to specify to model it with the simplified pvwatts equations.
>
> This would allow the model to handle other types of modules that don’t match the three predefined technologies, and would somewhat “future-proof” the model since it’s always possible to override the built-in options as the technology evolves.

janinefreeman, 2020-08-05:
A couple points on this:
- we updated the default panel characteristics in the 2020.2.29 release with version pvwattsv7, with significant input from industry. These new coefficients are shown in help, although we still have to complete an updated technical manual.
- the capability to input one's own module characteristics is feature of the detailed PV model within SAM, and has historically been one of our differentiating factors between the two models. That type of capability may or may not be outside of the scope of PVWatts.