NatLabRockies/ssc #457: Improve the backtracking routine for single-axis trackers

kanderso-nrel, 2020-10-11:
**Is your feature request related to a problem? Please describe.**
I see two ways to improve the single-axis backtracking routine implemented in `lib_irradproc.cpp`:
1) Remove the iterative "loop until there isn't shading" approach in favor of a closed-form equation.  The iterative approach overshoots by up to a degree and is unnecessary because analytical equations exist for bactracking.
2) Add the option to accommodate a "cross-axis" slope, for instance the slope between rows of a tracker array with N-S axes on an E-W slope.  If I understand correctly, SAM does not consider the cross-axis component of array slope in its SAT code.

**Describe the solution you'd like**
I'd like to implement the backtracking method described here: https://www.nrel.gov/docs/fy20osti/76626.pdf

Because the equations in that report are closed-form and can accommodate a ground slope that is non-parallel with the tracker axes, using that approach in SSC would address both of the above points.

**Describe alternatives you've considered**
None

**Additional context**
I think the current SSC self-shading code can be dramatically simplified using the approach in that report, and if the SAM team agrees, I would implement that as well.  I am happy to contribute code, tests, and documentation where appropriate, but will probably ask for guidance on matching the existing SSC style as I don't do much C++ and haven't contributed to SSC before.

cc @mikofski

mjprilliman, 2020-10-13:
@dguittet Do you think @kanderso-nrel  can start working on this or should we wait for Janine?

dguittet, 2020-10-14:
I think this looks really useful! I'm unable to download the technical report from nrel.gov at the moment-- do you have a copy @kanderso-nrel ? I'm curious how is the variation in the ground slope parametrized in your model?

We'd be very happy to have code, tests and documentation contributed, and to provide guidance about the SSC structure for the algorithm and then the integration of that algorithm into the SSC pv models. What can we do to help?

Our contribution guidelines are https://github.com/NREL/SAM/wiki/Contribution-Policy

dguittet, 2020-10-14:
[Slope-Aware Backtracking for Single-Axis Trackers.pdf](https://github.com/NREL/ssc/files/5379667/Slope-Aware.Backtracking.for.Single-Axis.Trackers.pdf)

Thanks @mjprilliman for sharing.

kanderso-nrel, 2020-10-14:
Strange, my browser can't get it from the NREL server either, but cURL downloads it no problem.  Thanks for rehosting it.  The report explains the approach better than I can in a github comment, but the short of it is that an additional parameter is needed that represents the component of ground slope that is perpendicular to the tracker axis (what I've been calling "cross-axis slope").  Note that the assumption is that the ground slope is uniform -- no rolling terrain.

My plan has really only considered `shared/lib_irradproc.cpp` in any depth: 
- Add a parameter to the `incidence` and `backtrack` functions for the cross-axis slope angle
- Modify the `backtrack` function to use the approach in TR/76626
- Modify the `shadeFraction1x` function to use the approach in TR/76626, haven't looked at this in depth but I think this would replace a few large functions with one small function.  Note that this change isn't necessary for the backtracking change, so it could be left as-is if we want to put it off for later or never. 

I will probably need some guidance on these, but I can take a first pass at it so we have something to build from:
- Add tests that cover the new code paths & functionality
- Because this involves adding an entirely new parameter, I guess the `_cm_vtab_pvsamv1` table in `cmod_pvsamv1.cpp` will need new rows.  Maybe new `subarrayN_cross_axis_slope` inputs, maybe with a better name?
- Set up the plumbing in `cmod_pvsamv1.cpp` to pass around the cross-axis slope parameter as needed.
- Same for other models (e.g. pvwatts?)

If this seems like I'm heading in the right direction, I'll open a PR with a first cut and we can iterate from there.

mjprilliman, 2020-10-15:
@kanderso-nrel @dguittet  After discussion we have decided that this work should be split into two parts. The backtracking and shadeFraction1x function improvements would be an improvement over the iterative method currently used in lib_irradproc. The best starting point would be to modify these functions without adding the cross axis-slope input (can you use a value of 0 for now?) so that we can have that in ssc sooner. The introduction of the new `subarrayN_cross_axis_slope` input is more complicated because it requires UI changes and changes to the var tables. We would also like to take the time to set up this input in a way that would be accessible to our other tracker options.

With that, you can definitely branch off of develop and start working whenever you are available to. You can reach out to me with any questions on working in SSC or general questions.

kanderso-nrel, 2020-10-15:
Thanks for the info!  I do have a branch that adds the relevant fields to `Subarray_IO` and `Irradiance_IO` but I am happy to start with the more basic implementation like you suggest.  I'll open a PR shortly.

janinefreeman, 2021-10-08:
Closing this issue since it was addressed in a PR (thanks Kevin!) and a more targeted issue #495 conveys the remaining piece needed here.

closed 2021-10-08 by janinefreeman