NatLabRockies/ssc #472: Default battery thermal doesn't work for sub-hourly simulations

brtietz, 2020-10-28:
**Describe the bug**
When attempting to run a detailed pv-battery simulation with a 30 minute weather file, I get the following error:

`exec fail(battery): Environment temperature input length must equal number of weather file records
Simulation pvsamv1 failed :exec fail(battery): Environment temperature input length must equal number of weather file records
`

This is with "Enter single fixed temperature" on the battery thermal page.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a PV battery simulation
2. Select a subhourly weather file
3. Run simulation
4. See error

**Expected behavior**
The default battery parameter (enter single fixed temperature) should be able to handle any weather file

brtietz, 2020-10-28:
This is new to the 2021 release (develop), in 2020.2.29 r3 this problem does not exist.

dguittet, 2020-10-28:
Currently in develop, `batt_room_temperature_celsius` must have length == nrec. I think this is unnecessary and could have length of 1 as well.

brtietz, 2020-10-30:
Closed by https://github.com/NREL/SAM/pull/418

closed 2020-10-30 by brtietz