NatLabRockies/ssc #494: Leap year in non-annual PV simulations

dguittet, 2020-12-04:
**Describe the bug**
Given the new functionality of non-annual simulations, the PV models use time stamp information directly. In particular, the hour of the year is calculated for a given timestamp. However, this is currently throwing an error when the data includes Feb 29.

**Expected behavior**
Would like to be able to calculate a correct hour of year regardless of when the leap day appears

@MRossol @grantbuster

MRossol, 2020-12-04:
welcome back @janinefreeman ;-)

dguittet, 2020-12-30:
Looking through the usage of the `hour_of_day` function, it seems to act as an indexer, particularly for the battery, shading_factor_calculator, adjustment factors and lifetime losses. 

- For the battery, which is not applicable with non-annual sims, the hour_of_day is used as an indexer as well.

- For the shading model, adjustment factors, and lifetime losses, using `hour_of_year` as an indexer assumes that the adjustment factors have been set up on an annual basis. For instance, if the weather data is a single time step but is really hour 4000 of the year, the time-series adjustment factors require that an array of length at least 4000 is provided.

   If I'm reading the code right, using these 3 features in the PV models doesn't preclude non-annual sims (whereas battery does). However, there is some disconnect in the length of data expected for non-annual sims: for weather, you can give 1 timestep, but for everything else, you must give the full 8760 * steps_per_hour. 

That makes it hard to generalize what `hour_of_year` is supposed to do.

 - If we want `hour_of_year` to be the index of the weather file instead, that would remove any dependency on leap days. If all the arrays for the four models' time-series parameters above are all the same length as the weather data, then this would work because the indices are aligned. However, this is not the case.
 - If there's 2/29 in the data but not 12/31 and it's an annual simulation, then `hour_of_year` could use the weatherfile's leap year flag to calculate the correct index.
 - If there's 2/29 in the data and it's not an annual simulation, then we could assume the shading, adjustment factors, and lifetime loss arrays have provided a slot for that leap day. But then if there's both 2/29 and 12/31 in the data, then we'll need to query the weather file to get a calculated index of 8760-8784. However, the time-series inputs to shading, adjustment factors, lifetime losses, etc, all assume 8760 * timestep length, so we'll end up with indices 0 - 24 at best and seg faults at worst.

I can think of two potential ways forward:

 - I think the most rigorous would be to make sure all the input time-series are for the same hours and minutes, so if there's a single day hourly sim, then all the other arrays should be 24 length. However, that's a rather large fix. 
 - Something we could do to fix the leap day issue is use the leap year flag so that 2/29 can return the correct index; then in the case of a non-8760 simulation with both 2/29 and 12/31, something will have to be done for 12/31.

janinefreeman, 2021-01-05:
I agree that it would make most sense to make sure all time series inputs are consistent. However, I think there may be cases when transferring data from another tool (particularly for shading info) where only an 8760 is available, and we would still want to interoperate with that. I think the assumption in that case is that Feb 29 is a repeat of Feb 28 as far as data retrieval goes. I could modify hour_of_year to fit that assumption, which could serve as a short-term fix to this problem- thoughts?

janinefreeman, 2021-01-29:
@MRossol this is running an 8760 that includes feb 29 but not dec 31 in pvsamv1, and assigning it as weather data? When was the last version that this worked? Thanks!

MRossol, 2021-01-29:
@janinefreeman e would LOVE to just be able to run the full 8784, but if not then the desired approach from ReEDS/PLEXOS is to drop dec 31.  Things broke after NREL-PySAM v2.1.4.

janinefreeman, 2021-01-29:
Thanks! With what I'm doing now, you should be able to run 8784, but you wouldn't get any annual outputs at the moment. Do you just use hourly outputs right now?

MRossol, 2021-01-29:
No, we use both :-(.  @grantbuster we have quite a few use-cases where we don't save the profiles correct?

janinefreeman, 2021-01-29:
Mmm ok. That's a bit larger of a fix. Let me investigate and see what I can come up with...

grantbuster, 2021-01-29:
Correct, we would want to be able to retrieve anything available in the SDK output variables list, with the profiles including the leap day. We do use quite a lot of your annual means.

janinefreeman, 2021-01-29:
Ok, helpful input. Thanks guys!

PR #537 (2021-02-24): Enable leap day to be included in PV simulations using weather data
This pull request enables February 29 to be included in a detailed PV simulation when inputting weather via the weather data variable. It also enables annual outputs for an 8760 simulation that includes Feb 29 but not Dec 31, specifically for our reV/REeDS folks (looking at you @grantbuster @mrossol ). Fixes #494 . It does NOT enable handling 8784 data, which is still on our wishlist.

After merging into patch, can we ask either @grantbuster or @mrossol to test a new ssc library to make sure this works for you? If so, let us k [... cut]
closed 2021-08-17 by janinefreeman