NatLabRockies/ssc #526: Fuel cell battery variable OM costs not calculated for production

mjprilliman, 2021-02-02:
**Describe the bug**
In the Fuel Cell-PV-Battery model, the variable production cost in $/kWh is input for batteries (om_production1) and fuel cell (om_production2). These values do not appear to be multiplied by the production anywhere in the financial models before being applied in the financial metric calculations. For example in cmod_singleowner.cpp:

Initialization: 
[image]

PV production costs are calculated, but not for battery or fuel cell: 
[image]

Included in operating expenses without any adjustments: 
[image]

Expected behavior: 
Production costs should be adjusted by multiplying by CF_energy_net or CF_energy_net_without_battery for matching units ($ rather than $/kWh).

mjprilliman, 2021-02-02:
@cpaulgilman @sjanzou My mistake. These inputs are not available in the UI so they are likely not intended to be used. Should they be added as part of the LCOS work?

sjanzou, 2021-02-02:
@mjprilliman, yes they should be added  but we should also clean them up and use a more descriptive naming convention.

sjanzou, 2021-02-02:
Also, as you noted, the production (variable) o and m costs should be multiplied by an output of system generation - either kWh or a sum of the power production at a timestep multiplied by the variable cost input.

mjprilliman, 2021-08-03:
This was fixed in #596. Closing for now

closed 2021-08-03 by mjprilliman