NatLabRockies/ssc #529: Clean up kWh outputs for two-meter case

brtietz, 2021-02-05:
The outputs “electricity to grid” and “electricity from grid” are just “gen when negative” and “gen when positive.” It seems like with buy all / sell all, both of these numbers could be positive at the same time since there are two meters. Is that a change that we want to make?

Figure out where these can be re-computed based on the utility rate. It feels weird to have the performance model be aware of the rate, but it might be appropriate for the two meters case?

Be sure to test with more systems than just PV.

dguittet, 2021-02-08:
This is about `year1_hourly_system_to_grid` (Electricity to grid) in utility rate, correct? And not `system_to_grid` (Electricity to grid from system) in battery? 

`revenue_gen` (Electricity to grid) also has the same label.

I agree the technology model shouldn't have to know about the compensation mechanism.

Perhaps we could clarify using "Net"?

brtietz, 2021-02-08:
We were looking at either the hourly or lifetime outputs when we wrote this issue, not annual. I think if we change one we should change both, it would be very confusing to have the hourly outputs be different from the lifetime outputs.

dguittet, 2021-02-09:
Right, `year1_hourly_system_to_grid` is an array, as is `system_to_grid`. Except the first one is a utility rate output and the second is a battery output. These are both computed using the net power transfer, as is `revenue_gen`, so the electricity to grid and electricity from grid vectors are complementary. But if the utility rate ones should be not necessarily be complementary when using Net Billing, then the battery ones will have to be changed also for consistency.

brtietz, 2021-02-09:
Yes - for buy all / sell all (option 4, not net billing), sales and purchases can both happen in the same timestep. We can currently use system power generated and a load variable to see gross purchases and gross sales and then multiply them by the buy and sell rates, but Paul and I were thinking the above variables might be confusing for that compensation option.

I think we have the option of changing the outputs for buy all / sell all, or adding some notes to the help to steer users in the right direction.

cpaulgilman, 2021-02-10:
Somewhat related issue over on the SAM repo: https://github.com/NREL/SAM/issues/377