NatLabRockies/ssc #539: Buy all / sell all incorrectly assigns tier to inverter night time losses

brtietz, 2021-02-18:
After https://github.com/NREL/ssc/pull/538, the kWh w/o system are allocated correctly for buy all / sell all when using kWh/kW rates:

[image]

However, the inverter night time losses are allocated incorrectly:

[image]

Figure out how to allocate the night time losses to the highest tier. I think this has something to do with the code around cmod_utility_rate5 line 1071, but should check with @sjanzou first.

sjanzou, 2021-02-19:
@brtietz, does the inverter loss get added to the load appropriately before the utility rate compute module? What are the charges and tier breakdown with no load (and only inverter loss)? If this is just bookkeeping around line 1071, then the logic should check all the energy use at line 1123 against the upper bound for that tier and roll any excess to the next tier... so, definitely a bug but hard to tell if just bookkeeping or actual charge implications...

cpaulgilman, 2021-02-19:
I just had a user support discussion that might be relevant to this regarding the effect of transformer losses (which scale with inverter size) on demand charges for months when the peak time step is at night. In this context, I just want to be sure that when we say "inverter loss" we mean the total nighttime loss.

[pv-inverter-transformer-losses-test.zip](https://github.com/NREL/ssc/files/6011510/pv-inverter-transformer-losses-test.zip)

brtietz, 2021-02-19:
@sjanzou Based on my work for https://github.com/NREL/SAM/issues/370 (and soon to be work on https://github.com/NREL/SAM/issues/532) the inverter/transformer nighttime losses are negative numbers in gen, and not included in the load. My hunch is that this will affect the charges in addition to the bookkeeping. 

Good call on testing without load. Here's what happens to that same system:

[image]

The tier structure is:
[image]

and peak demand based on the nighttime losses alone is 0.339 kW, so that should be split 79.8 kWh in tier 2, and the rest in tier 6.

brtietz, 2021-03-03:
This is closely related to https://github.com/NREL/ssc/issues/546, so move this to the fall release so all of the work happens on develop.

brtietz, 2021-06-14:
The line numbers have shifted since this issue was first filed. Here's an updated permalink to save time in the future: https://github.com/NREL/ssc/blob/ec110c1a19e69e40d603c32655aab7341ff39519/ssc/cmod_utilityrate5.cpp#L1183

sjanzou, 2021-11-20:
@brtietz, where is the test .sam file for the 

> @sjanzou Based on my work for [NREL/SAM#370](https://github.com/NREL/SAM/issues/370) (and soon to be work on [NREL/SAM#532](https://github.com/NREL/SAM/issues/532)) the inverter/transformer nighttime losses are negative numbers in gen, and not included in the load. My hunch is that this will affect the charges in addition to the bookkeeping.
> 
> Good call on testing without load. Here's what happens to that same system:
> 
> [image]
> 
> The tier structure is: [image]
> 
> and peak demand based on the nighttime losses alone is 0.339 kW, so that should be split 79.8 kWh in tier 2, and the rest in tier 6.

@brtietz, where is the .sam project file used to generate these tables? Thanks!

brtietz, 2021-11-23:
@sjanzou The file used to generate the above is here:
[billing_demand_demo.zip](https://github.com/NREL/ssc/files/7590338/billing_demand_demo.zip) - the "ratchets off" case is probably the most useful for this issue.

However, for this issue I recommend testing with kWh tiers and nice round numbers. Here's an example for that: 
[evenly_distributed_tiers_test.zip](https://github.com/NREL/ssc/files/7590341/evenly_distributed_tiers_test.zip)

sjanzou, 2021-11-28:
With pull request #673 , the inverter loss is moved to the highest tier that has not reached upper bound.
Test file 
[evenly_distributed_tiers_test_sj.zip](https://github.com/NREL/ssc/files/7613509/evenly_distributed_tiers_test_sj.zip)

Results before pull request
[image]

Results after pull request
[image]

closed 2021-11-29 by sjanzou