NatLabRockies/ssc #546: Fix Accounting for Inverter Operating Losses in LCOE/Gen

janinefreeman, 2021-03-02:
Currently, inverter operating losses (day and nighttime) are subtracted off of `gen`, which is the energy used in the numerator of the LCOE. However, clarification from a solar/wind developer indicates that the energy required for inverter operating losses should actually be counted as an operating cost, and that energy should NOT be subtracted from `gen`. The operating cost can be calculated either at the PPA price (if no retail rate is specified), OR at the utility rate. (We'll still need to figure out what to do for behind-the-meter systems)

From the developer: "For the cash flow models she uses during the development phase to determine a PPA price to use in proposals for solar projects, she gets a single annual kWh estimate from her engineering team to account for total inverter consumption (day and nighttime), and then applies either the PPA price or a retail rate to estimate the operating cost. Because the kWh value is relatively small for solar projects, some utilities don't bother billing at the retail rate and treat it as negative generation, but other utilities do charge a retail rate. Retail rates are typically commercial rates with an energy charge and demand charge. On the wind side, site consumption is a larger fraction of the total system output, so this value has a bigger impact on the cash flow."

brtietz, 2021-03-03:
Related issues: https://github.com/NREL/ssc/issues/539 https://github.com/NREL/SAM/issues/532

brtietz, 2021-05-12:
Here's an interesting way to quantify this from patch 2:

This is the PV-Battery Single Owner default for patch 2. The battery only charges from PV in this case.

[image]

When I flip on the retail rate, the cost of the inverter night-time losses goes down (from $0.09 to $0.06), so the NPV goes up (which makes sense) but so does the LCOE (which doesn’t make sense):

[image]
 
The reason for the LCOE increase is increased OPEX, as discussed above.

When we've solved this bug, the decreased cost of night-time losses should be reflected in the LCOE and the NPV.

PR #651 (2021-11-16): LCOE calculations with parasitic load costs in numerator, Energy sales in denominator
-Updated LCOE calculation in all financial models
-Parasitic load operating cost based on PPA price or retail rate for FOM
-Parasitic load operating cost based on hourly prices for BTM
-Variable generation O&M calculated based on energy of value (FOM-sales, BTM-to grid or to load) not net energy
-LCOS updated with same LCOE methods for calculation of cost of grid charging
-Closes #546 outside of larger powerflow changes
closed 2021-11-16 by mjprilliman