NatLabRockies/ssc #558: Large battery storage systems segfault

ajcressman, 2021-03-22:
Using MATLAB, I was able to create a project (~100MW AC) with two sizes of battery storage - (1) ~10MWh of capacity, and (2) ~100MWh of capacity.  SSC build 250 (tried 252 as well, but same issue).

The 10MWh project runs fine, but the 100MWh project segfaults.  It seems that the crash is related to `dispatch_automatic_front_of_meter_t::dispatch` (this is not the first time I've encountered a crash in this function; at least one other time it was due to a different battery chemistry profile).  I should note that the latest PySAM also segfaults with these inputs.

I've attached a JSON of the working project, of the failing project, the solar resource file that was used, and the MATLAB crash report: [storage_segfault.zip](https://github.com/NREL/ssc/files/6178855/storage_segfault.zip)

Ideally I would like assistance in what values are giving me problems, but at least I would hope that future releases of SSC have more checks and return more error messages, rather than segfaulting entirely.  Thank you in advance!

dguittet, 2021-04-01:
I tried to reproduce your error with the new PySAM version 2.2.2 but wasn't able to. Could you try the following script?

```
import json
import PySAM.Pvsamv1 as pv
import PySAM
print(PySAM.__version__)

params_file = "/Users/dguittet/SAM-Dev/support/storage_segfault/not_working.json"
weather_file = "/Users/dguittet/SAM-Dev/support/storage_segfault/49857911773588960TYA.CSV"

m = pv.new()

vals = json.load(open(params_file, "r"))

for k, v in vals.items():
    try:
        if k[0] == '6':
            m.value('six' + k[1:], v)
        else:
            m.value(k, v)
    except:
        for a in m.__dir__():
            if a == "BatteryCell":
                x = 0
            try:
                for b in m.__getattribute__(a).__dir__():
                    if b.lower() == k:
                        m.value(b, v)
                        break
            except:
                # print(k, v)
                pass
        # except:
        #     print(k, v)

m.SolarResource.solar_resource_file = weather_file

# these values didn't get loaded correctly from json so do it manually
m.InverterDatasheet.inv_tdc_ds = [[1, 25, 0],]
m.AdjustmentFactors.constant = 0
m.AdjustmentFactors.dc_constant = 0
m.value("ppa_price_input", (0, ))
m.BatteryCell.batt_voltage_matrix = [[0, 0],]

m.execute()
```

janinefreeman, 2021-10-08:
@ajcressman Looks like this issue may have been fixed with the related issue #562 . Closing for now, please reopen if you're still seeing the segfault.

closed 2021-10-08 by janinefreeman