NatLabRockies/ssc #562: Segmentation fault in irrad::getBackSurfaceIrradiances with bifacial modules

tkelman, 2021-04-07:
**Describe the bug**
Originally noticed in a Matlab R2019b simulation (on both Windows and Linux), but managed to reproduce in pysam (latest 2.2.2 release, ubuntu 20.04) with a script similar to #558.

**To Reproduce**
Steps to reproduce the behavior:
`python crash.py`
[crash.zip](https://github.com/NREL/ssc/files/6272746/crash.zip)

**Expected behavior**
Successful run or descriptive error message, not a segmentation fault.

**Desktop (please complete the following information):**
 - OS: ubuntu 20.04
 - Version 2.2.2

**Additional context**
Backtrace from `gdb --args python crash.py` (the build from pip wheel doesn't look like it has detailed debuginfo included)
```
(gdb) r
Starting program: /home/tkelman/code/envs/pysam-latest/bin/python crash.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
2.2.2

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff579c5a9 in irrad::getBackSurfaceIrradiances(double, double, double, double, double, double, std::vector<double, std::allocator<double> >, std::vector<double, std::allocator<double> >, std::vector<double, std::allocator<double> >, std::vector<double, std::allocator<double> >&, double&) ()
   from /home/tkelman/code/envs/pysam-latest/lib/python3.8/site-packages/PySAM/libssc.so
(gdb) bt
#0  0x00007ffff579c5a9 in irrad::getBackSurfaceIrradiances(double, double, double, double, double, double, std::vector<double, std::allocator<double> >, std::vector<double, std::allocator<double> >, std::vector<double, std::allocator<double> >, std::vector<double, std::allocator<double> >&, double&) ()
   from /home/tkelman/code/envs/pysam-latest/lib/python3.8/site-packages/PySAM/libssc.so
#1  0x00007ffff579e88f in irrad::calc_rear_side(double, double, double) ()
   from /home/tkelman/code/envs/pysam-latest/lib/python3.8/site-packages/PySAM/libssc.so
#2  0x00007ffff50b9c5d in cm_pvsamv1::exec() ()
   from /home/tkelman/code/envs/pysam-latest/lib/python3.8/site-packages/PySAM/libssc.so
#3  0x00007ffff52bb2b3 in compute_module::compute(handler_interface*, var_table*) ()
   from /home/tkelman/code/envs/pysam-latest/lib/python3.8/site-packages/PySAM/libssc.so
#4  0x00007ffff52daf47 in ssc_module_exec_with_handler ()
   from /home/tkelman/code/envs/pysam-latest/lib/python3.8/site-packages/PySAM/libssc.so
#5  0x00007ffff6393630 in SAM_module_exec ()
   from /home/tkelman/code/envs/pysam-latest/lib/python3.8/site-packages/PySAM/libSAM_api.so
#6  0x00007ffff741c507 in Pvsamv1_execute (self=0x7ffff77f4a20, args=<optimized out>) at modules/Pvsamv1.c:14251
#7  0x00000000005f305a in PyCFunction_Call ()
#8  0x00000000005f3446 in _PyObject_MakeTpCall ()
#9  0x000000000056f600 in _PyEval_EvalFrameDefault ()
#10 0x000000000056822a in _PyEval_EvalCodeWithName ()
#11 0x000000000068c1e7 in PyEval_EvalCode ()
#12 0x000000000067d5a1 in ?? ()
#13 0x000000000067d61f in ?? ()
#14 0x000000000067d6db in PyRun_FileExFlags ()
#15 0x000000000067da6e in PyRun_SimpleFileExFlags ()
#16 0x00000000006b6132 in Py_RunMain ()
#17 0x00000000006b64bd in Py_BytesMain ()
#18 0x00007ffff7deb0b3 in __libc_start_main (main=0x4eec80 <main>, argc=2, argv=0x7fffffffdca8, init=<optimized out>,
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdc98) at ../csu/libc-start.c:308
#19 0x00000000005f927e in _start ()
```

dguittet, 2021-04-24:
The issue here is that the sixpar_area value was 0, leading to issues in the selfshading module such as a width and length of 0, which was then passed on to the bifacial calculations, leading to NaNs. I'll add some error checks along the process to make this easier to debug next time.

tkelman, 2021-04-24:
Great, that's definitely a bug in some of my upstream code sending zero areas, I should have looked through the parameters more carefully. Error checking will be helpful.

PR #567 (2021-04-30): Add error checks for when PV module dimensions are 0
When a PV module has an area of 0, the calculated dimensions of the module lead to erroneous behavior in the self-shading and bifacial calculations. Catch this error in pvsamv1 and irradproc.

Close #562
commit 79bf2052 (2021-05-06): Change PV open circuit current labels to fix #562
closed 2021-05-02 by tkelman