NatLabRockies/ssc #565: voltage_table_t::calculate_max_discharge_w breaks dispatch constraints

brtietz, 2021-04-21:
Lead acid batteries using the voltage table can export to grid, since the dispatch current can be greater than the load. The root cause of the issue is that voltage_table_t::calculate_max_discharge_w specifies unreasonably high output currents, and this is the last constraint that is enforced.

Unclear whether this is a units issue with qmax, or something else. A brute force solution would be to enforce dod is >0 and <100 in the function.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a PV-Battery residential case with a lead acid battery that uses the voltage table
2. Run the model with any dispatch
3. Note that (a) the SOC goes well below the SOC limits specified on the dispatch page, and (b) the electricity from the battery is greater than the load, meaning this constraint is violating other constraints.

**Expected behavior**
voltage_table_t::calculate_max_discharge_w should respect charge limits specified in the dispatch functions. 

**Screenshots**
Peak shaving:

[image]

Specified limits:

[image]

**Desktop (please complete the following information):**
 - Version 2020.11.29 r1 - problem exists on patch, and the code is the same on develop.

brtietz, 2021-04-21:
It looks like the issue is that the voltages in the default lead acid voltage table are bank voltages, not cell voltages. Three recommendations for resolution:

(1) Update the Lead Acid default to a lower table voltage (should be ~2 V @ 50%)
(2) Do add the "brute force" dod constraints in the discharge function above.
(3) Consider a warning for the voltage table values: for example if all of the values are strictly above or strictly below the cell voltage, prompt the user to adjust the cell voltage.

dguittet, 2021-04-22:
Can you link to where `voltage_table_t::calculate_max_discharge_w` is leading to the over-dispatch? This function is meant to calculate the maximum power available to be discharged. Regardless of whether there is an issue in bank vs cell voltages which should definitely be fixed, this function is meant to be used by the dispatcher to determine max limits, and not vice versa ( no dependency of voltage model on dispatch models ). Perhaps a new function is required such as "calculate_current_for_SOC" or something.

The voltage of the table should be used in calculating the number of cells in series so if it were provided as a bank voltage, then there should just be one cell in series. In the leadacid defaults I'm seeing the Vnom_default = 2 instead of 12 so that's probably causing the issue. An error message of "voltages in table exceed Vnom_default * 2" should help.

brtietz, 2021-04-22:
This is where the problematic current makes it into the dispatch code: https://github.com/NREL/ssc/blob/bc4cfcf446370cf5691eab7534b5d14641b154ea/shared/lib_battery_dispatch.cpp#L152

The dispatch SOC constraints should adjust qmax here: https://github.com/NREL/ssc/blob/bc4cfcf446370cf5691eab7534b5d14641b154ea/shared/lib_battery.cpp#L527

but because of the issue with the nominal voltage and the number of cells, the values of A and B result in a dod of 1200% here: https://github.com/NREL/ssc/blob/7fe77525bb52fc8373698a93edb0d28035e790d0/shared/lib_battery_voltage.cpp#L192

I don't think we need a new function, but I do think worth limiting dod to 0% and 100% to prevent garbage values from coming out of this function.

cpaulgilman, 2021-04-22:
Just a note that the voltage table might also be used for a flow battery or other battery type, so the solution should not be limited to lead acid chemistries.

brtietz, 2021-04-27:
PR 568 is a partial solution for patch 2. A full solution will take more time and be implemented as a part of https://github.com/NREL/ssc/issues/569.

closed 2021-05-07 by brtietz