NatLabRockies/ssc #581: Snow modeling assumptions for true-tracking vs back-tracking in PVWatts7 appear to differ unexpectedly

cpr-chas, 2021-05-27:
The snow model library makes an assumption on the number of strings per slant height based on the tracking type:
File: cmod_pvwatts7.cpp

            if (pv.type == ONE_AXIS)
                pv.nmody = 1; // e.g. Nextracker or ArrayTechnologies single portrait
            else
                pv.nmody = 2; // typical fixed 2 up portrait

Based on this logic, a backtracking system (type == ONE_AXIS_BACKTRACKING) would get the 2-up version.  I'm guessing this was unintentional and that all single-axis tracker types should get nmody of 1.

I'd be happy to contribute a fix and test coverage after the desired behavior is confirmed.
I haven't done any specific testing yet - just found this while learning about snow modeling.

janinefreeman, 2021-05-27:
@cpr-chas You're absolutely right, the intended behavior is what you describe and not what's happening in the code. If you have a chance to contribute a fix and test, that would be fantastic! Thanks! :)

cpr-chas, 2021-05-27:
Sure.  The change is obviously simple, but it might take me a day or two to figure out the test side.

janinefreeman, 2021-05-27:
Let us know if you need any support on that side, we're happy to help!

janinefreeman, 2021-08-17:
Hi @cpr-chas ! Just checking in on this as we start to corral issues for a fall release. Thanks!

PR #643 (2021-11-05): Merge in CPR's Snow Model Tests and PVWatts Bug Fix
Branch created from pull request #582 from @cpr-chas with some updates. Fixes #581 and adds several tests. Thank you @cpr-chas!!
closed 2021-11-05 by janinefreeman