NatLabRockies/ssc #599: Merchant Plant negative cleared capacities?

janinefreeman, 2021-08-03:
What should the merchant plant do with negative cleared capacity? It could:
a) not allow negative cleared capacity and throw an error
b) assume that negative cleared capacity means you're charging storage- in which case it could assume that 1) you're buying power at the current cleared price, or 2) you're buying power at a retail rate.

This needs some further investigation.

brtietz, 2022-01-21:
It looks like (b) is what's assumed currently - this is with cleared capacity = 100% of generation:

[image]

However, this is not correct, at I had selected "use retail electricity rates" for this model, so the merchant plant needs to do some of the same massaging of gen that single owner does currently.

Additionally, the line item for "electricity purchases" runs to be nearly the same as revenue when "use ppa or market prices" is chosen:

[image]

This is for a case where the battery only charges from PV, so I'm concerned a $/MWh is being interpreted as a $/kWh somewhere in this configuration.

janinefreeman, 2022-05-24:
Decision at SAM meeting is to ensure that option b) 2) is what the behavior is. The assumption is that your "negative cleared capacity" is actually electricity that needs to be purchased for your system.

brtietz, 2022-05-26:
Testing so far has identified the following problems:

- If using market rates for electricity purchases, $/MWh needs to be converted to $/kWh (a fix for this is up on branch ssc_599_negative_cleared_capacity, but the additional issues should be resolved before a PR)
- If using market rates for electricity purchases, cleared capacity is not accounted for, the full grid usage is multiplied by the price
- "Energy market revenue" still includes the negative values of gen, so regardless of the purchase source the negative values are double counted in revenue and operating expenses right now.

The first and third have obvious fixes. For the second, if a project draws energy from the grid do we:

(a) Throw an error if the draw does not match the cleared capacity? OR
(b) Charge the project for everything it draws regardless of the cleared capacity? OR
(c) Charge the project for the cleared capacity regardless of what it draws?

Additionally, does cleared capacity matter if the project is purchasing at a retail rate?

Note that the oft-delayed https://github.com/NREL/SAM/issues/181 means it might be a while before batteries could account for charging limits if we choose (a).

janinefreeman, 2022-05-27:
Hmm. Thanks for the detailed investigation. For the second bullet, it does seem that "cleared capacity" should match the actual draw from the grid. I'd be tempted to go with (a)... or possibly (b), but throwing warnings for each timestep to indicate that it's happening. Thoughts from @cpaulgilman ?

brtietz, 2022-05-27:
One clarifying question with (b) - should any "extra" energy be purchased from the energy market by default, or whichever market has the largest cleared capacity? I'm a fan of the warnings at each timestep idea.

Also, for the third bullet point above, I was planning on splitting the positive and negative values of gen (/cleared capacities) so that the grid draw shows up as an operating expense. This should help make the LCOE consistent with single owner.

cpaulgilman, 2022-05-27:
I associate "capacity" with generation, so I'm not sure what it means in the context of consumption. The purpose of the cleared capacity requirement is to guarantee that the system generates a minimum amount of power. I don't think it would apply to consumption, but that's just my intuitive sense. For a project that uses market prices for electricity purchases, I think it would be reasonable to go with (b) and use the energy market price as the purchase price along with the time step warnings.

closed 2022-05-31 by brtietz